moveit/moveit #1716: Files without software license agreement information

jliukkonen, 2019-11-08:
I scanned the MoveIt master branch as a part of testing scancode-toolkit (https://github.com/nexB/scancode-toolkit) and upon closer inspection noticed that a dozen or so files are missing software license agreement information. I have listed the files and their paths below. 

Files without license:
- [ ] /moveit/moveit_planners/trajopt/include/trajopt_interface/problem_description.h
- [ ] /moveit/moveit_planners/trajopt/include/trajopt_interface/trajopt_planning_context.h
- [ ] /moveit/moveit_planners/chomp/chomp_interface/src/chomp_planning_context.cpp
- [ ] /moveit/moveit_planners/sbpl/core/sbpl_interface/src/bfs3d/search.cpp
- [ ] /moveit/moveit_planners/trajopt/src/kinematic_terms.cpp
- [ ] /moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_detector_fcl_plugin_loader.h
- [ ] /moveit/moveit_core/collision_detection_fcl/src/collision_detector_fcl_plugin_loader.cpp
- [ ] /moveit/moveit_planners/trajopt/include/trajopt_interface/kinematic_terms.h
- [ ] /moveit/moveit_planners/trajopt/src/trajopt_planning_context.cpp
- [ ] /moveit/moveit_planners/sbpl/core/sbpl_interface/src/bfs3d/bfs_3d.cpp
- [ ] /moveit/moveit_planners/trajopt/test/trajectory_test.cpp
- [ ] /moveit/moveit_experimental/moveit_jog_arm/src/teleop_examples/spacenav_to_twist.cpp

Files with conflicting licence reports (check the license wording):

- [ ] /moveit/moveit_planners/trajopt/src/problem_description.cpp'

davetcoleman, 2019-11-08:
Thanks @jliukkonen , I've just added usernames to groups of files, of the original or related author who I think can fix this. If I just tagged you: could you create a PR adding an open source license (preferably BSD) so we can close this issue?

pombredanne, 2019-11-08:
@jliukkonen glad to see that scancode-toolkit could be of some help :heart: ... and if you have anything not detected correctly, that's a bug you can send my way. 

Since you are updating some license details, may I also suggest you have a look into https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/process/license-rules.rst (I helped a bit there) and https://reuse.software/ as a possibly to cleanup how things are documented license-wise? 
(and as a side note, you could go farther and adopt similar convention for catkin/ros package.xml manifests. FWIW I am working to do something similar for Python packages: https://github.com/pombredanne/spdx-pypi-pep/blob/4614112f35a937550ce6234799804a65eb279580/pep-9999.rst )

jliukkonen, 2019-11-09:
@pombredanne thanks, it's a pretty handy tool. :smiley:  One thing I noticed while doing this, is that I couldn't find an option to mark and list files with missing license information. So I wrote a tiny Python script to post-process the output file (I chose .csv for easy parsing). Is there a way to detect missing licenses already? I'd like to know how to get list of those as well if it is possible already although the script I wrote already helps me to find them just fine.

The options I used, and which did not produce the list of files without license was:
`./scancode --license-score 50 -n 4 -l <moveit_master_branch> --csv <output>`

j-petit, 2019-11-10:
The two `collision_detector_fcl_plugin_loader.h` and the cpp respectively were added by @BryceStevenWilley. I made a PR to add the license, see #1720.

ommmid, 2019-11-11:
Added the license in [this commit](https://github.com/ros-planning/moveit/pull/1626/commits/9d65345a7098ce94388f36b19e7216ef4058b1c3)

pombredanne, 2019-11-13:
@jliukkonen you wrote:

> it's a pretty handy tool. smiley One thing I noticed while doing this, is that I couldn't find an option to mark and list files with missing license information. 

If you use the JSON output, that would be `files` entries that have a `type` of `file` and that have an empty `license_expressions` array.

> So I wrote a tiny Python script to post-process the output file (I chose .csv for easy parsing). Is there a way to detect missing licenses already? I'd like to know how to get list of those as well if it is possible already although the script I wrote already helps me to find them just fine.

Feel free to share the script so I can review it if you want.  

> The options I used, and which did not produce the list of files without license was:
./scancode --license-score 50 -n 4 -l <moveit_master_branch> --csv <output>

I see what you are looking for now! a list of the files with no license detected at all (or with a  score less than 50/100) and only that so you can get a punch list of sorts. A ticket is welcomed if I can coerce you in entering one at https://github.com/nexB/scancode-toolkit/issues/ ... that should be easy enough as a simple plugin!