moveit/moveit #1717: CHOMP output trajectory has 0.0 time_to_start duration

truhoang-ubtrobot, 2019-11-08:
### Description
This is more of a bug I think.

I'm trying to chain OMPL and CHOMP planner according to the documentation and keep getting the error:
```
[ERROR] [1573174064.950117312, 33.165000000]: Trajectory message contains waypoints that are not strictly increasing in time.
```

So I went digging into the source code and found that when CHOMP update the points in it's plan (call to `addSuffixWaypoint()`), it uses `0.0` for the duration (in chomp_planner.cpp):
```cpp
auto result = std::make_shared<robot_trajectory::RobotTrajectory>(planning_scene->getRobotModel(), req.group_name);
  // fill in the entire trajectory
  for (size_t i = 0; i < trajectory.getNumPoints(); i++)
  {
    const Eigen::MatrixXd::RowXpr source = trajectory.getTrajectoryPoint(i);
    auto state = std::make_shared<robot_state::RobotState>(start_state);
    size_t joint_index = 0;
    for (const robot_state::JointModel* jm : result->getGroup()->getActiveJointModels())
    {
      assert(jm->getVariableCount() == 1);
      state->setVariablePosition(jm->getFirstVariableIndex(), source[joint_index++]);
    }
    result->addSuffixWayPoint(state, 0.0);
  }
```

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source <master> build

I have a way to fix it in my application code but I think implementing a solution to source would be better. Any thoughts or suggestion would be much appreciated.

rhaschke, 2019-11-08:
MoveIt augments time parameterization to geometric trajectories via a specific planning_adapter plugin.
That is, the planner (here chomp) doesn't need to provide the timing yet.
Looks like you need to fix the order of your planning_adapters to have chomp being applied before adding timing.

truhoang-ubtrobot, 2019-11-08:
I edited the `ompl_planning_pipeline.launch` to call chomp before the time parameterization (according to the [tutorial](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/chomp_planner/chomp_planner_tutorial.html#using-ompl-as-a-pre-processor-for-chomp)) so shouldn't the time be already added? I printed the motion plan to the terminal and it's `0.0` for every waypoint.

```xml
<arg name="planning_adapters" value="
           default_planner_request_adapters/FixWorkspaceBounds                
           default_planner_request_adapters/FixStartStateBounds           
           default_planner_request_adapters/FixStartStateCollision
           default_planner_request_adapters/FixStartStatePathConstraints
           chomp/OptimizerAdapter
           default_planner_request_adapters/AddTimeParameterization
           " />
```

Artificially adding the time makes the plan acceptable to the robot.

closed 2020-02-06 by truhoang-ubtrobot

edetleon, 2020-03-27:
> I edited the `ompl_planning_pipeline.launch` to call chomp before the time parameterization (according to the [tutorial](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/chomp_planner/chomp_planner_tutorial.html#using-ompl-as-a-pre-processor-for-chomp)) so shouldn't the time be already added? I printed the motion plan to the terminal and it's `0.0` for every waypoint.
> 
> ```
> <arg name="planning_adapters" value="
>            default_planner_request_adapters/FixWorkspaceBounds                
>            default_planner_request_adapters/FixStartStateBounds           
>            default_planner_request_adapters/FixStartStateCollision
>            default_planner_request_adapters/FixStartStatePathConstraints
>            chomp/OptimizerAdapter
>            default_planner_request_adapters/AddTimeParameterization
>            " />
> ```
> 
> Artificially adding the time makes the plan acceptable to the robot.

Hello truhoang-ubtrobot,

Did you finally figure this out ? I want to extract velocity curves and I am running into the same issue. 

Regards

truhoang-ubtrobot, 2020-03-27:
@edetleon 
for my application, after I got the motion plan, I love through each waypoint and manually add the time. Since `my_plan[0]` has to start at `t=0` and I know my robot requires each waypoint to be exactly 10ms apart, I add it the following way:

```
dt = 0.0;
for (each waypoint):
    waypoint.time_from_start = dt;
    dt += 0.01;
```

edetleon, 2020-03-27:
> @edetleon
> for my application, after I got the motion plan, I love through each waypoint and manually add the time. Since `my_plan[0]` has to start at `t=0` and I know my robot requires each waypoint [...]

Thank you for your quick answer ! So you added the time yourself, but could you visualize different velocities in Rviz ?

truhoang-ubtrobot, 2020-03-27:
> Thank you for your quick answer ! So you added the time yourself, but could you visualize different velocities in Rviz ?

I'm not sure what RViz requires. From my understanding the AddTimeParameterization plug-in does a similar thing. It may also discretize the waypoints by linear interpolation so that it takes exactly `dt` (for my case 10ms) for the arm to move there. 

I don't see why you wouldn't be able to visualize it with Rviz.