moveit/moveit #1721: WMD19 - Prepare issues

v4hn, 2019-11-11:
World MoveIt Day is around the corner and almost all open issues with the `moveit day candidate` label are older than a year (or were added by me).

If you have *any* unresolved issue in or around MoveIt in mind that you would like to see resolved/ improved, but that is not yet documented as an issue here, please take the time to write down the essential information required to reproduce it and have someone write a patch for it!

If the issue can be resolved within a day, add the `moveit day candidate` label so people can tackle the issue next week.

This goes especially to all official maintainers
@davetcoleman @rhaschke @130s @IanTheEngineer @gavanderhoorn @jrgnicho @bmagyar @mlautman @jonbinney @henningkayser @BryceStevenWilley @mamoll

felixvd, 2019-11-18:
Could you add the label to #1730 , #1556 and #1065 ? I can't do it myself. Thanks!

rhaschke, 2019-11-18:
> Could you add the label to #1730 , #1556 and #1065 ?

Done.

v4hn, 2019-11-25:
over for this year.

closed 2019-11-25 by v4hn