moveit/moveit #1722: Python 3 support

v4hn, 2019-11-11:
The MoveIt Python API is not yet fully functional with Python 3 and scarcely tested.
As one example, `rosrun moveit_commander moveit_commander_cmdline.py` does not provide a usable shell.

It only makes sense to look into this issue, if you have previous experience with ROS on Python 3.
This is not the default configuration of ROS and you will need to build many standard packages manually to attempt it.

Briancbn, 2019-11-20:
`moveit_ros_planning_interface` package needs editing to generate the python3 APIs

I tried to compile the whole moveit using python3 by running by setting the catkin config:
```
catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so -DCMAKE_BUILD_TYPE=Release
```
Then I also manually add in the python3 directories into the PYTHONPATH
```
export PYTHONPATH=/home/rosi/Documents/moveit_ws/devel/lib/python3/dist-packages:$PYTHONPATH
```
I edit the first line of the `moveit_commander_cmdline.py` to `#!/usr/bin/env python3`run the `rosrun moveit_commander moveit_commander_cmdline.py` command. It gives me a segfault, it comes from importing the packages from `moveit_ros_planning_interface`

130s, 2020-05-30:
“I’ll work on this for WMD 2020”. However I can't guarantee full-day commit so I'd appreciate company.
I've done some Python3 conversion from Python2.

lorepieri8, 2021-03-31:
Any progress on this?

v4hn, 2021-03-31:
Please try to use the APIs in ROS noetic and report errors if you should find any.
The state of the python3 migration is unclear to me, but with ROS noetic releases it is supposed to work now.

lorepieri8, 2021-03-31:
Where I can find the python API? 

Besides https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html where I can I look at? The source code?

felixvd, 2021-04-02:
There is also the auto-generated code API documentation:  
[MoveGroupInterface](http://docs.ros.org/en/api/moveit_python/html/classmoveit__python_1_1planning__scene__interface_1_1PlanningSceneInterface.html)  
http://docs.ros.org/en/api/moveit_python/html/classmoveit__python_1_1move__group__interface_1_1MoveGroupInterface.html
[PlanningSceneInterface](http://docs.ros.org/en/api/moveit_commander/html/classmoveit__commander_1_1planning__scene__interface_1_1PlanningSceneInterface.html)
[RobotCommander](http://docs.ros.org/en/api/moveit_commander/html/classmoveit__commander_1_1robot_1_1RobotCommander.html)  
[Another PlanningSceneInterface]??(http://docs.ros.org/en/api/moveit_python/html/classmoveit__python_1_1planning__scene__interface_1_1PlanningSceneInterface.html)