moveit/moveit #1726: Moveit 0.9.17 update for Kinetic can not solve previously solvable points

penglongxiang, 2019-11-13:
### Description

It seems after updating moveit to version 0.9.17 in Kinetic, the planning performance is not as good as before (version 0.9.15).  

### My environment
* ROS Distro: [Kinetic]  
* OS Version: Ubuntu 16.04  
* Binary build  
* If binary, which release version? (upgrade from 0.9.15 to 0.9.17)  
* If source, which git commit or tag?  

### Steps to reproduce
1. update moveit related debian packages to the latest version. If possible, prepare another moveit 0.9.15 version for comparison test.     
2. Download and follow the build steps in [xArm ROS packages](https://github.com/xArm-Developer/xarm_ros) .  
3. After successful build, follow the guide in [xarm_planner](https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_planner) package and execute following command to start the robot simulation and plan for a target trajectory:
```bash
roslaunch xarm_planner xarm_planner_rviz_sim.launch robot_dof:=6
```
and then in another terminal:  
```bash
rosservice call xarm_pose_plan 'target: [[0.322, 0.2, 0.22], [0.707, -0.139, 0.657, 0.22]]'
```

### Expected behaviour
Tell us what should happen:
In previous 0.9.15 version, this target can be solved and executed with no problem on this xArm model. The output in 0.9.15 is:  
[ INFO] [1573617585.194456521]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1573617585.195452107]: Planner configuration 'xarm6' will use planner 'geometric::RRT'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1573617585.195676417]: RRT: Starting planning with 1 states already in datastructure
[ INFO] [1573617585.231398735]: RRT: Created 237 states
[ INFO] [1573617585.231456112]: Solution found in 0.035841 seconds
[ INFO] [1573617585.246975805]: SimpleSetup: Path simplification took 0.002825 seconds and changed from 6 to 2 states

### Actual behaviour
Tell us what happens instead  
While we did another test on moveit version 0.9.17, with the same xarm_ros package and environment. The above target point can not be solved anymore, the output is now:  
[ INFO] [1573030328.933719950]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1573030328.958714122]: Planner configuration 'xarm6' will use planner 'geometric::RRT'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1573030328.959087852]: RRT: Starting planning with 1 states already in datastructure
[ INFO] [1573030333.959101703]: RRT: Created 34066 states
[ INFO] [1573030333.959155341]: No solution found after 5.000197 seconds
[ INFO] [1573030333.966949212]: Unable to solve the planning problem  
 
The implementation code of the planning service is [here](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_planner/src/xarm_simple_planner.cpp):  just using the basic move_group_interface plan() function to solve the trajectory and nothing complicated.  

This issue is reported by one of xArm application developer, since this position is critical to them and it worked well previously, so would you please suggest some potential cause of this issue? What has
been changed in the new version?   

Thanks in advance!

v4hn, 2019-11-13:
Looking through the commits between the versions, I see no obvious candidate that might trigger this.
I quickly ran your example with MoveIt's master branch as well as an unreleased rather recent version of OMPL and cannot reproduce the problem.

Planning succeeds as expected:
[image]

Are you sure this is directly related to MoveIt?
I could imagine OMPL and FCL might also have had new package versions with your updates and either could be responsible.

Possible Steps to debug this:
- Use `rqt_logger_level` to set `move_group`'s `ros.moveit_planners_ompl` logger to `Debug.
 This might provide further insights.
- Try to plan to the joint pose you expect it to find (just c&p the one from 0.9.15)
  This simplifies the problem and ensures that the IK solver is not involved.
- Build MoveIt from source with version 0.9.17, see if you can reproduce the problem.
- Add the relevant versions of OMPL and FCL to the workspace (you need `catkin build` for this)

If you can reproduce the problem with MoveIt built from source, please bisect to find the faulty commit.

penglongxiang, 2019-11-14:
@v4hn Thank you very much for the detailed suggestion, so ompl and moveit are independent right? I will follow your suggested steps and see which part leads to this situation.

penglongxiang, 2019-11-18:
hi @v4hn , I download the master branch of moveit source code, as well as ompl and fcl and built them with 'catkin build'. Fcl and ompl were built successfully,  but the building of 'moveit_core' failed. The error messages are as follows:  
```bash
Errors     << moveit_core:make /home/jason/catkin_ws/logs/moveit_core/build.make.002.log                                                  
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp: In function ‘void moveit::core::{anonymous}::_attachedBodyToMsg(const moveit::core::AttachedBody&, moveit_msgs::AttachedCollisionObject&)’:
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:219:14: error: ‘moveit_msgs::AttachedCollisionObject_<std::allocator<void> >::_object_type {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
   aco.object.subframe_names.clear();
              ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:220:14: error: ‘moveit_msgs::AttachedCollisionObject_<std::allocator<void> >::_object_type {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’
   aco.object.subframe_poses.clear();
              ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:223:16: error: ‘moveit_msgs::AttachedCollisionObject_<std::allocator<void> >::_object_type {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
     aco.object.subframe_names.push_back(frame_pair.first);
                ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:226:16: error: ‘moveit_msgs::AttachedCollisionObject_<std::allocator<void> >::_object_type {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’
     aco.object.subframe_poses.push_back(pose);
                ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp: In function ‘void moveit::core::{anonymous}::_msgToAttachedBody(const moveit::core::Transforms*, const AttachedCollisionObject&, moveit::core::RobotState&)’:
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:255:22: error: ‘const _object_type {aka const struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’
       if (aco.object.subframe_poses.size() != aco.object.subframe_names.size())
                      ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:255:58: error: ‘const _object_type {aka const struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
       if (aco.object.subframe_poses.size() != aco.object.subframe_names.size())
                                                          ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:303:48: error: ‘const _object_type {aka const struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’
         for (std::size_t i = 0; i < aco.object.subframe_poses.size(); ++i)
                                                ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:306:35: error: ‘const _object_type {aka const struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’
           tf2::fromMsg(aco.object.subframe_poses[i], p);
                                   ^
/home/jason/catkin_ws/src/moveit/moveit_core/robot_state/src/conversions.cpp:307:41: error: ‘const _object_type {aka const struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
           std::string name = aco.object.subframe_names[i];
                                         ^
make[2]: *** [robot_state/CMakeFiles/moveit_robot_state.dir/src/conversions.cpp.o] Error 1
make[1]: *** [robot_state/CMakeFiles/moveit_robot_state.dir/all] Error 2
make: *** [all] Error 2  
```

Do you have any idea what went wrong? It was built on ROS Kinetic and ubuntu 16.04.  
  
BTW, I tried the above mentioned planning problem on ROS Melodic with the latest binary package updates, it can be solved successfully. But it will still fail on Kinetic.

v4hn, 2019-11-18:
You also have to include `moveit_msgs` with the appropriate branch.