moveit/moveit #1727: How to use the MoveGroupCapability clearOctomap plugin

rickstaa, 2019-11-13:
### Description

I would like to clear the octomap after I performed a camera pose estimation. I first tried doing this using the move_group API directly, but I couldn't get this to work. I then saw moveit also supplies a plugin that can be used to clear the Octomap [see the documentation](https://moveit.ros.org/documentation/plugins/). I created the following plugin_loader.cpp to load this plugin:

```
#include "ros/ros.h"

#include <pluginlib/class_loader.h>
#include <moveit/move_group/move_group_capability.h>

int main(int argc, char** argv)
{

  // Initialize node
  ros::init(argc, argv, "clear_cotomap_loader");

  pluginlib::ClassLoader<move_group::MoveGroupCapability> move_Group_capability_loader("moveit_ros_move_group", "move_group::MoveGroupCapability");

  try
  {
    boost::shared_ptr<move_group::MoveGroupCapability> clear_octomap = move_Group_capability_loader.createInstance("move_group/ClearOctomapService");
    clear_octomap->initialize();
    ROS_INFO("The clear octomap plugin was loaded successfully.");
  }
  catch(pluginlib::PluginlibException& ex)
  {
    ROS_ERROR("The plugin failed to load for some reason. Error: %s", ex.what());
  }

  ros::spin();
  return 0;
}
```

Unfortunately, the `move_group/clearOctomap` service is not showing up when executing `rossservice list` command. 

I added the following lines to my `CMAKEList.txt` file:

```
add_executable(clear_octomap_loader src/clear_octomap_loader.cpp)
target_link_libraries(
  clear_octomap_loader
  ${catkin_LIBRARIES}
  ${Boost_LIBRARIES}
)
add_dependencies(
  clear_octomap_loader
  ${${PROJECT_NAME}_EXPORTED_TARGETS} # Msgs, srv, actions of your package
  ${catkin_EXPORTED_TARGETS}  # Msg, svr, action srv from other packages
)
```

### Your environment
* ROS Distro: Melodic
* OS Version:  Ubuntu 18.04
* Binary: v1.0.2

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
I expected the `move_group/clearOctomap service to show up when running the `rosservice list` command.

### Actual behaviour
It did not.

v4hn, 2019-11-14:
Hi Rick,

with any recent MoveIt version, the capability is automatically loaded and provides the service `/clear_octomap`.
If you call this service, the Octomap maintained in the `move_group` node will be cleared.

rickstaa, 2019-11-14:
@v4hn Thanks for your response. You are right thanks that's a lot easier!

closed 2019-11-14 by rickstaa