moveit/moveit #1728: Stop octomap update

LHJ1098826475, 2019-11-14:
### Description

I want to control physical robot to grasp object by using moveit-ros, but due to the octomap automatically updating, object I need to grasp also be updated as a part of octomap, so robot can't touch it. I want to know how to stop octomap update when I use moveit to come true trajectory-planning. Moreover, moveit of ros is lib file, how can I change the file. 

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

v4hn, 2019-11-14:
(1) If you add an appropriate collision object, the octomap will be cleared in the vicinity of the object. You still have to allow the collision with the collision object of course.

(2) If you really want to disable octomap updates, the easiest way to do so is externally by writing a node that forwards point clouds to the topic `move_group` subscribes to only when required.

The mentioned problem is no issue with MoveIt itself, so I will close the issue here.

closed 2019-11-14 by v4hn

LHJ1098826475, 2019-11-14:
I can write a node to publish topic with octomap, but now octomap info will be considered by moveit when I use moveit to trajectory planning. So I want to stop octomap update in moveit lib file.

v4hn, 2019-11-14:
I'm not sure what you actually want, but if you also want to clear the octomap (to remove potential collisions from the planning scene), call the `clear_octomap` ros service.