moveit/moveit #1730: Test camera calibration

felixvd, 2019-11-15:
If you are a new user participating in WMD2019 and have a robot with a camera (either robot-mounted or scene-mounted), try using the changes in #1070 to calibrate the camera and report your experience.

Advanced users can fix bugs and submit code, but even sharing just feedback is valuable, so this is a good topic that even multiple new users can tackle.

You will need to compile the branch from source to test this feature.

RoboticsYY, 2019-11-18:
I’ll work on the bug fix for hand-eye calibration tool for WMD 2019.

felixvd, 2019-11-18:
Great. Just to reiterate, we also need lots of people to test the feature, so this issue is still open to anyone who wants to try it out!