moveit/moveit #1731: Validate existing MoveIt tutorials documentation [Melodic]

cardboardcode, 2019-11-15:
# Why?

Similar to Issue #645, this issue thread is created to facilitate checks for existing Getting Started or Advanced MoveIt Tutorials specifically for the version **ROS Morenia Melodic**

While the issue linked is similar, it seems it is specific to the **ROS Kinetic Kame** version of the MoveIt Tutorials. As @davetcoleman mentioned in [this comment](https://github.com/ros-planning/moveit/issues/645#issuecomment-433091876), more testing is needed for Melodic.

# Purpose?
Highlight as much spelling errors, code syntax errors spotted in the documentation as possible. 
The assignee (probably me) will collate all the needed changes and make a pull request. 

# Link
https://docs.ros.org/melodic/api/moveit_tutorials/html/index.html

cardboardcode, 2019-11-20:
## MoveIt Tutorials Sections `Validated`
* [CHOMP Planner](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/chomp_planner/chomp_planner_tutorial.html)
* [STOMP Planner](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html) 
* [Planning Adapter Tutorial](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/planning_adapters/planning_adapters_tutorial.html) 
* [Joystick Control Teleoperation](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/joystick_control_teleoperation/joystick_control_teleoperation_tutorial.html)
* [Benchmarking](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/benchmarking/benchmarking_tutorial.html)
* [Integrations/Unit Tests](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/tests/tests_tutorial.html)

## Required Edits
The section below lists what are the potential edits required to **correct** and/or **improve** the documentation for the [MoveIt Melodic Tutorials](https://docs.ros.org/melodic/api/moveit_tutorials/html/index.html).

Each edit is corresponded to the specific **Section** or **Section header**. 

### [CHOMP Planner](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/chomp_planner/chomp_planner_tutorial.html)

1. Add full stop (.) at the end of the statement description **smoothness_cost_weight** under Section header, **[Tweaking some of the parameters for CHOMP](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/chomp_planner/chomp_planner_tutorial.html#tweaking-some-of-the-parameters-for-chomp)**.
**[Rationale]** - Sentence structure error. 

2. Place in bullet point for the **unused/commented parameters** under Section header, **[Tweaking some of the parameters for CHOMP](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/chomp_planner/chomp_planner_tutorial.html#tweaking-some-of-the-parameters-for-chomp)**. 
**[Rationale]** - Clearer format for easy reading.

3. Bold the parameter variables under Section header, **[Tweaking some of the parameters for CHOMP](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/chomp_planner/chomp_planner_tutorial.html#tweaking-some-of-the-parameters-for-chomp)**.
**[Rationale]** - Clearer format for easy reading.

### [STOMP Planner](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html) 

1.  Put in bullet point for the **Strengths of STOMP** under Section Header, **[STOMP Planner](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html#stomp-planner)**.
**[Rationale]** - Clearer format for easy reading.

2. Fix `more info` hyperlink under Section Header, **[STOMP Planner](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html#stomp-planner)**.
**[Rationale]** - For providing proper link to compliment tutorial information.

3. Bold the parameter variables under Section header, **[Tweaking some of the parameters for STOMP](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html#tweaking-some-of-the-parameters-for-stomp)**.
**[Rationale]** - Clearer format for easy reading.

4. Remove `STOMP` after first line in content under Section Header, **[Difference between plans obtained by STOMP, CHOMP and OMPL](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html#difference-between-plans-obtained-by-stomp-chomp-and-ompl)**
**[Rationale]** - Trivial typo error. 

5. Clarification of phrase **within reasonable times** under Section Header, **[STOMP Planner](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html#stomp-planner)**
**[Rationale]** - Would provide a more concrete image of what level of proficiency STOMP operates in. 

`The rest of the sections validated do not require further edits.`

sheikspeare, 2019-11-20:
**MoveIt Tutorials Sections Validated**

_Perception Pipeline Tutorial_

-> [Detecting and Adding Object](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html#detecting-and-adding-object-as-collision-object)
1. Replace rviz with RViz 
[Rationale] - Be consistent with naming conventions.

2. Transform cache issue - is this updated and resolved by @Ridhwanluthra & @MohmadAyman ?

Have gone through Perception pipeline until OMPL Planner, seems to be fine.

rhaschke, 2019-11-20:
Thanks @cardboardcode and @sheikspeare for validating the tutorials.
Could you, please, directly file PRs with the proposed changes? The tutorials reside in [`doc/*/*.rst` files](https://github.com/ros-planning/moveit_tutorials/tree/melodic-devel/doc) in the repo https://github.com/ros-planning/moveit_tutorials. You can even edit those files directly on github (and preview your changes). Thanks a lot!

cardboardcode, 2019-11-20:
@rhaschke  
Hi Mr Haschke, noted. :smile:  

While I wait for the rest of my team to validate the rest of the MoveIt tutorials, I will compile the edit suggestions here and make an overall Pull Request (PR). 

Will update again once all the validation has been done.

davetcoleman, 2019-11-20:
Thanks for validating these! 

Let's also make sure whatever changes get merged into melodic, also get merged into the ``master`` branch where active development is.

ngwk1, 2019-11-22:
**MoveIt Tutorials Sections Validated**

```
Getting Started
MoveIt Quickstart in RViz
Move Group C++ Interface
```
-> [Getting Started](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html)
1. Opened an issue and submitted a pull request [#425](https://github.com/ros-planning/moveit_tutorials/pull/425). 

-> Moveit Quickstart in RViz and Move Group C++ Interface
1. Seems fine.

lw21, 2019-11-22:
MoveIt Tutorial Sections Validated:

- [Planning Scene ROS API](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/planning_scene_ros_api/planning_scene_ros_api_tutorial.html)
- [Motion Planning API](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/motion_planning_api/motion_planning_api_tutorial.html)

- [Motion Planning Pipeline](https://docs.ros.org/melodic/api/moveit_tutorials/html/doc/motion_planning_pipeline/motion_planning_pipeline_tutorial.html)

The documentation is fine. No fault found in this sections.

EdmundNg1998, 2019-11-22:
MoveIt Tutorial Sections Validated

- [Pick and Place](https://ros-planning.github.io/moveit_tutorials/doc/pick_place/pick_place_tutorial.html)

- [Moveit Grasps](https://ros-planning.github.io/moveit_tutorials/doc/moveit_grasps/moveit_grasps_tutorial.html)

- [Moveit Task Constructor](https://ros-planning.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html)

- [Moveit Setup Assistant](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html)

- [URDF and SRDF](https://ros-planning.github.io/moveit_tutorials/doc/urdf_srdf/urdf_srdf_tutorial.html)

- [Low Level Controllers](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html)

The documentation is alright. No errors found in these sections.