moveit/moveit #1732: Display CollisionObjects' frames in Rviz

felixvd, 2019-11-15:
#1439 added subframes to CollisionObjects, but no visualization was added at the time. The frames and subframes of CollisionObjects should be introspectable in Rviz, however.

This issue most likely involves changing the Rviz plugin to display frames of a PlanningScene. This will be useful for introspecting them in the MTC as well.

This is an issue for WMD2019, and sort of a duplicate of #1122.

rhaschke, 2019-11-15:
Why do you clutter the issues with duplicates (of #1122)? The other thread is way more informative.
I added the label WMD to the other issue and close this one here.

closed 2019-11-15 by rhaschke

felixvd, 2019-11-15:
Because it seemed appropriate to summarize the thread for users who are scanning for simple WMD issues, it's a clearer title, and I figured it won't hurt as long as they are clearly linked. Sorry?

rhaschke, 2019-11-15:
I adapted the title of #1122.