moveit/moveit #1736: Fix setup assistant

v4hn, 2019-11-15:
With a number of recent changes to the code base, new bugs were introduced that make the setup assistant hard to use in some cases.

This is a meta-issue to collect individual issues.

To contribute to this issue, go through the motions to create a new moveit_config package for a urdf/xacro you have around or look at your existing package through `roslaunch <robot>_moveit_config setup_assistant.launch` and try to perform reasonable adjustments.
@henningkayser and I encountered a number of issues with the custom [panda_env](https://github.com/henningkayser/ROS-Industrial_EU_Fall19_MoveIt) robot we used in a workshop recently).

- [ ] When loading the urdf / the package the robot does not appear on the right side until one tries to add a Robot Pose. This break visualization of every step before that.
- [ ] I seem to recall adding two named robot poses reset some of the joints of the first one (???)
- [ ] The `ros_controllers.yaml` file deals with `ros_control` controller configuration, but somehow got mixed up with the `controllers.yaml`, which describes static ROS interfaces for sending trajectories
- I seem to recall some complains that some configuration files were overwritten unconditionally.
- ... 

No guarantee on any of these issue, I listed those off the top of my head. :man_shrugging:

gavanderhoorn, 2019-11-15:
> I seem to recall some complains that some configuration files were overwritten unconditionally.

Perhaps #1680 and #1679?

rhaschke, 2019-11-18:
> When loading the urdf / the package the robot does not appear on the right side

I performed a `git bisect` for this issue and figured out that the breaking commit is b0f01b2dc79f99f5650933834c8e50905bdbe168, which was part of #1625.