moveit/moveit #1743: Python3 move_commander_cmdline error in Melodic with _moveit_roscpp_initializer

Briancbn, 2019-11-20:
### Description
https://github.com/ros-planning/moveit/issues/1722

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04.3 LTS
* Source build
* master branch commit 5b44334bcc5666ebf615b63ebd6f80ebb800b905

### Steps to reproduce
I install the python3 components through running
```
sudo apt-get install python3-pip python3-yaml
sudo pip3 install rospkg catkin_pkg

```
I tried to compile the whole moveit using python3 by running by setting the catkin config:
```
catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so -DCMAKE_BUILD_TYPE=Release
```
Compile succeeded
Then I also manually add in the python3 directories into the PYTHONPATH
```
export PYTHONPATH=/home/rosi/Documents/moveit_ws/devel/lib/python3/dist-packages:$PYTHONPATH
```
I edit the first line of the `moveit_commander_cmdline.py` to `#!/usr/bin/env python3`

run the `rosrun moveit_commander moveit_commander_cmdline.py` command. It gives me an error, it comes from importing the packages from `moveit_ros_planning_interface`, it seems that it cannot find the py_binding_tools _roscpp_initializer 
```
Traceback (most recent call last):
  File "/home/rosi/Documents/moveit_ws/src/moveit/moveit_commander/bin/moveit_commander_cmdline.py", line 16, in <module>
    from moveit_commander import MoveGroupCommandInterpreter, MoveGroupInfoLevel, roscpp_initialize, roscpp_shutdown
  File "/home/rosi/Documents/moveit_ws/devel/lib/python3/dist-packages/moveit_commander/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 2, in <module>
  File "/home/rosi/Documents/moveit_ws/src/moveit/moveit_commander/src/moveit_commander/roscpp_initializer.py", line 35, in <module>
    from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: /home/rosi/Documents/moveit_ws/devel/lib/python3/dist-packages/moveit_ros_planning_interface/_moveit_roscpp_initializer.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE
```

### Expected behaviour
No error running as expected

### Actual behaviour
Error message shown as above

v4hn, 2019-11-22:
The observed issue might actually be an artifact because the rest of ROS, that MoveIt depends on was not build with python3.

To validate, I would very much recommend to build *an entire ROS distribution* from scratch with python set to python3.

junyaoshi, 2020-01-02:
@v4hn Thanks for your response! I ran into the same issue. How should I build an entire ROS distribution from scratch with python set to python3? Any pointers?

Briancbn, 2020-01-03:
@junyaoshi I still haven't had time to continue doing it, but this [guide](https://www.miguelalonsojr.com/blog/robotics/ros/python3/2019/08/20/ros-melodic-python-3-build.html) will help. Also remember to check your boost version, it seems that lower boost version will revert back to using python