moveit/moveit #1747: MoveIt Release Process Owners

davetcoleman, 2019-11-20:
I want to clarify here who is leading releases for the MoveIt project. We have a pressing need to [re-release panda_moveit_config](https://github.com/ros-planning/moveit/pull/1626#issuecomment-533918473).

For years @130s has been on top of it, but more recently @rhaschke has lead releases I believe? PickNik would also like to help with the release process, as @JafarAbdi has become an expert on Debians recently. 

@rhaschke are you ok with Jafar doing this release?

rhaschke, 2019-11-20:
I'm happy if Jafar can release panda_moveit_config.
For MoveIt I'm more hesitant because that needs much more experience and attention.

rhaschke, 2019-11-20:
Leaving this open until @JafarAbdi has prepared the release for panda_moveit_config.

JafarAbdi, 2019-11-20:
We should first port the new changes from [melodic-devel](https://github.com/ros-planning/panda_moveit_config/commits/melodic-devel) to [kinetic-devel](https://github.com/ros-planning/panda_moveit_config/commits/kinetic-devel), otherwise, Travis will fail again on kinetic

davetcoleman, 2019-11-20:
> We should first port the new changes from melodic-devel to kinetic-devel, otherwise, Travis will fail again on kinetic

The trajopt changes only target ``master``, and I'm nervous about breaking kinetic. I'm starting to think we need to drop kinetic support for master...

Also: I don't see a ``panda_moveit_config-release`` repo, have we ever released it or has it been Franka Emika?

JafarAbdi, 2019-11-20:
Oh my bad, I forget that the Travis kinetic test run on the master branch, 

The MoveIt release repos are hosted in `ros-gbp` -- this's the one for panda_moveit_config https://github.com/ros-gbp/panda_moveit_config-release

davetcoleman, 2019-11-20:
You now have write access to both panda_moveit_config and panda_moveit_config-release

davetcoleman, 2020-05-06:
Following up on this topic, beyond just panda_moveit_config...

@rhaschke [just released](https://github.com/ros-planning/moveit/issues/1990) the latest melodic debians, which is very appreciated.

We had an internal meeting at PickNik on how we can best support the release process going forward. We would like to commit to doing this quaterly, for melodic, noetic, and ROS 2. We came up with this process.

For now we are asking @mlautman to do the steps on behalf of PickNik that require write-access, which @JafarAbdi does not yet have.

- @mlautman Announce 2 weeks prior to release on Discourse that we are making one
  - Ask that nothing be merged in 1 week prior (code freeze) to melodic-devel
- Optional Step:
  - @rhaschke / @v4hn / whomever is motivated: Back port remaining fixes from master to melodic-devel
- @JafarAbdi Pre-release testing
  - Several run-time tests
    - Test Rviz Motion Planning Plugin
  - Run bloom pre-release test
- @mlautman Update CHANGELOGS / package.xml version numbers
- @mlautman Blooming / sending to ROS Build Farm
- @mlautman Announce the release whenever ROS buildfarm syncs

rhaschke, 2020-05-06:
@mlautman, as you will prepare the CHANGELOGs in the future: I continued @130s's tradition of classifying each commit as a feature, bugfix, or maintenance change. Usually, this also involves some reduction of verbosity (catkin_generate_changelog just collects all commit messages) or grouping. Have a look at some recent changelogs to get an idea of this.

davetcoleman, 2020-06-02:
I think this issue can be closed, lots of new release issues for specific version have popped up at this point, lead by @tylerjw

closed 2020-06-02 by davetcoleman