moveit/moveit #1767: LazyPRM dosn't create any edges when range parameter is too low

GabrielKoenig, 2019-11-22:
### Description
I'm using moveit and the ompl library from source. In my application I need a very small state space discretization (with RRTConnect my range parameter should be set to 0.1 due to narrow passages ). Motivation: I don't want that moveit interpolates too much between the states (otherwise the planner will never find a solution for narrow passage task). I want that around 90% of the states created are actually created by the OMPL planners. I tried to achieve a similar discretization with raodmaps planner with LazyPRM as with RRTConnect, but failed....  (thus I tried to investigate the issue using the general panda_moveit_config) 

### Your environment

* ROS Distro: melodic
* OS Version: Ubuntu 18.04
* Source build
* melodic-devel (9aaad286ae7e) with only some changes for visualization (see: https://github.com/GabrielKoenig/moveit_visualize_data) 

### Steps to reproduce
To investigate the influence of the different planner parameter I adapted the moveit code to visualize all sampled states (OMPL states and not the final interpolated ones from moveit) and all the edges. Steps to reproduce: 
1. Clone both my branches for moveit (https://github.com/GabrielKoenig/moveit) and the node to visualize the data (https://github.com/GabrielKoenig/moveit_visualize_data/tree/master) (you can use OMPL from source or binary just respect the changes you need to do for either one: https://moveit.ros.org/install/source/dependencies/)
2. You can try to run the panda_moveit_config demo.launch node (and add within RVIZ MotionPlanning and MarkerArray (subscribing to marker topic /rviz_visual_tools). 
4. Set the parameter "PlanningAttempts=1" (my visualization tool only works if we don't have any parallel plan running)
5. Create a motion plan request with the MotionPlanning  Display with the LazyPRM for the panda_arm without any obstacles with `range` parameter 1 and `longest_valid_segment_fraction` 0.001. 
6. Visualize the sampled states with roslaunch `moveit_visualize_data sampling_data_analysis.launch`
You can also run any panda_moveit tutorial and set the Planner to LazyPRM and the planning time to half an hour or longer. 

### Expected behaviour
Especially when dealing with narrow passages, I need e very small range parameter such that the edge distance is not too big (I don't want too much interpolation) I would expect that a robot (e.g. Panda) should find a solution for the LazyPRM with range 0.1. 1 or 2 with a planning time of half an hour and created 400'000 states. 

### Actual behaviour
In general I am aware, that the number of states sampled for PRM or LazyPRM is dependent on the planning time. To visualize the undesired edge creation I visualized all the valid states and edges after 5 seconds (I can not show the images for 30min because this would crash rviz with so many staes....). In all pictures, the longest_valid_segment_fraction is set to 0.001. The green dots represent the sampled states from OMPL visualized only as the position of the panda_link8 and the red lines show the edges created by OMPL. 

In figure 1) shows the desired discretization resp. distance between the states using RRTConnect (range 0.1). Similar connection behaviour I would like to achieve with the LazyPRM
[image]

no figure) When setting the range to 0.1 and using the LazyPRM, the planner never finds more than 2 states even giving the planner time for half an hour and creating over 400'000 states. 

In figure 2)  range is set to 1  (no solution, 16'818 states created, 132 valid vertices, 130 edges)
[image]

In figure 3) range is set to 2 (no solution, 1'795 states created, 11'702 valid vertices, 68'874 edges)
[image]

In figure 4) range is set to 5 (it findes a solution, 11'164 states created, 11'166 valid vertices, 109'064 edges) 
[image]

I think it is very unsuitable to have a planner that only works because moveit nows how to interpolate between the states which are far away from each other. A range parameter of 5 is just creates too much unnecessary edges... 

Questions: 
1) Am I right that in general (for all planners) the 'longest_valid_segment_fraction' is only used for the maximum allowed motion during collision checking (and has mainly an influence during interpolation) and has no influence on the distance function between two states that are connected through an edge. (only if the OMPL planner use the range parameter, the maximum connection length between two edges can be influenced) @BryceStevenWilley I saw you worked manly on this parameter in #869, am I right?  
2) What might be the reason, that LazyPRM doesn't create any vertices and edges even giving the planner half an hour to run? 
3) By the way: What do you think about my implementation to visualize the sampled states and edges from OMPL and visualize them through moveit? (I refer also to my question here: https://answers.ros.org/question/334475/how-to-display-planner-data-in-moveit/). I see that in moveit the functionality for visualizing the sampled states doesn't work (the functionality is commented out) 

Thanks for the help.

BryceStevenWilley, 2019-12-06:
Sorry for the lateness, I'll comment on what I know, although I'm not an expert on LazyPRM's behavior. 

> Am I right that in general (for all planners) the 'longest_valid_segment_fraction' is only used for the maximum allowed motion during collision checking

Yes, you are correct. It will make collision checking (and therefore planning as a whole) slower, but it shouldn't affect the length of edges in the planner at all. Like you mention, if it's too low, the planner will end up adding edges that are actually in collision.

> What might be the reason, that LazyPRM doesn't create any vertices and edges even giving the planner half an hour to run?

I'm not sure. It's very curious that there are so few vertices, my best guess is that maybe LazyPRM is removing singular vertices too soon? They could also simply not be showing up in the planner data that you are visualizing (given that a large portion of the LazyPRM graph is not collision checked, I don't know if you are filtering those unchecked vertices out).  

>  What do you think about my implementation to visualize the sampled states and edges from OMPL and visualize them through moveit?

I think it's great and would definitely be useful to other developers.  You should open a PR for the moveit side, but I'm not sure where the Visualizer code would go though.

GabrielKoenig, 2019-12-08:
Thanks for your reply. 

> Yes, you are correct. It will make collision checking (and therefore planning as a whole) slower, but it shouldn't affect the length of edges in the planner at all. Like you mention, if it's too low, the planner will end up adding edges that are actually in collision.

Thanks for the confirmation, but don't you mean if the maximum_longest_valid segment faction is set too **high**, then the planner will end up adding edges that are actually in collision? 

> I'm not sure. It's very curious that there are so few vertices, my best guess is that maybe LazyPRM is removing singular vertices too soon? They could also simply not be showing up in the planner [...]

Could be a reason... I don't know... @mamoll could it be that during this stage in LazyPRM https://github.com/ompl/ompl/blob/master/src/ompl/geometric/planners/prm/src/LazyPRM.cpp#L531-L535 too many vertices are deleted from the roadmap?  
 
> I think it's great and would definitely be useful to other developers. You should open a PR for the moveit side, but I'm not sure where the Visualizer code would go though.

Okey I will give it a try.

mamoll, 2019-12-08:
If the range is set to a small value, then initially LazyPRM will create lots of vertices but very few edges, because hardly any vertex will have any neighbors within the range cutoff. *Eventually*, once you have sampled "enough", vertices will have some neighbors within the range cutoff, and the roadmap becomes more connected. Does that explain the behavior you are seeing?