moveit/moveit #1774: How can I stop printing statements from moveit_commander Python?

Murtaza786-oss, 2019-11-25:
Hi, I have a code here made using moveit_commander and I am using RL for training a UR5. I obtain pose and joint angle values by running the moveit_commander interface and publishing those values in a rostopic so that I can subscribe it in my environment. But due to the training overload, the publishers seem to crash and the whole training process stops. 
My idea is to somehow incorporate the moveit_commander within the environment itself so that I can get rid of publishers and subscribers. The problem with moveit-commander is that whenever you initialze the moveit_cmmander interface and run the python code it starts printing these statements.

    ```
[ INFO] [1574631433.496141382]: Loading robot model 'ur5_r2'...
    [ WARN] [1574631433.497492371]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
    [ INFO] [1574631433.497565114]: No root/virtual joint specified in SRDF. Assuming fixed joint
    [ INFO] [1574631433.696682137, 7.780000000]: Loading robot model 'ur5_r2'...
    [ WARN] [1574631433.696776037, 7.782000000]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
    [ INFO] [1574631433.696798683, 7.782000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
```

I somehow want to edit this and stop printing these boiler statements. Is there a way to do this? because these statements will significantly slower the training process.

felixvd, 2019-11-25:
I feel like this should be on ROS Answers and I don't really see how print statements will significantly slow down the learning process, but it looks like you can achieve what you're thinking of by setting the [logging level](http://wiki.ros.org/roscpp/Overview/Logging) to ROS_ERROR or ROS_FATAL.

gavanderhoorn, 2019-11-25:
> I feel like this should be on ROS Answers [..]

it already is, this is a cross-post: [How can I stop printing statements from moveit_commander Python?](https://answers.ros.org/question/338451)

rhaschke, 2019-11-25:
Please follow the instructions on ROS answers to increase the logging level. Closing here.

closed 2019-11-25 by rhaschke