moveit/moveit #1775: Segfault in jog_arm

jschleicher, 2019-11-25:
### Description

jog_arm server segfaults in position mode.
Does the velocity array exist, if settings are `publish_joint_velocities: false`?

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build master fd1dba5d96491cf31b7b854e7da8fcd81796b314

### Steps to reproduce
1. clone https://github.com/PilzDE/pilz_teach
2. increase joint scale in `prbt_jog_settings.yaml:12`to max. `joint: 1.57`
3. `roslaunch prbt_jog_arm_support prbt_jog_arm_sim.launch`
4. Send twist command with 
```
rostopic pub -r 100 /jog_server/delta_jog_cmds geometry_msgs/TwistStamped "header: auto
twist:
  linear:
    x: 0.0
    y: 1.0
    z: -1.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```

### Expected behaviour
velocity limited to the maximum available

### Actual behaviour: Backtrace and Console output

Segfault:
```
[ INFO] [1574683431.721719141]: Received first command msg.
[ WARN] [1574683432.002629698]: /jog_server prbt_joint_5  close to a  velocity limit. Enforcing limit.

Thread 6 "jog_server" received signal SIGSEGV, Segmentation fault.
[...]
585	          new_jt_traj.points[0].velocities[c] = kinematic_state_->getJointVelocities(joint)[0];
(gdb) bt
#0  0x000055555558081c in moveit_jog_arm::JogCalcs::checkIfJointsWithinURDFBounds (this=this@entry=0x7fffe142a200, new_jt_traj=...)
    at /home/jschleicher/catkin_ws/src/moveit/moveit_experimental/moveit_jog_arm/src/jog_calcs.cpp:585
#1  0x0000555555585e8f in moveit_jog_arm::JogCalcs::cartesianJogCalcs (
    this=this@entry=0x7fffe142a200, cmd=..., shared_variables=..., 
    mutex=pthread_mutex_t = {...})
    at /home/jschleicher/catkin_ws/src/moveit/moveit_experimental/moveit_jog_arm/src/jog_calcs.cpp:321
#2  0x000055555558884f in moveit_jog_arm::JogCalcs::JogCalcs (
    this=0x7fffe142a200, parameters=..., shared_variables=..., 
    mutex=pthread_mutex_t = {...}, model_loader_ptr=...)
    at /home/jschleicher/catkin_ws/src/moveit/moveit_experimental/moveit_jog_arm/src/jog_calcs.cpp:154
#3  0x00005555555a1863 in moveit_jog_arm::JogROSInterface::startJogCalcThread (
    this=<optimized out>)
    at /home/jschleicher/catkin_ws/src/moveit/moveit_experimental/moveit_jog_arm/src/jog_ros_interface.cpp:163
```

AndyZe, 2020-01-20:
#1837 likely fixes this

JafarAbdi, 2021-02-03:
@jschleicher Could you please check if this still an issue .?

jschleicher, 2021-02-04:
Sorry, I cannot reproduce that. We're no longer using the pilz_teach repo and there have been too much changes to the moveit_servo config files to make the old config work. Thus closing here.

closed 2021-02-04 by jschleicher