moveit/moveit #1785: [ros_planning_interface] Undefined symbol on python module import with geometry2 and moveit installed from source

gleichdick, 2019-11-27:
### Description

According to the [Move Group Tutorial](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html), I tried to import the `moveit_commander.conversions` python module on an openSUSE source setup.
It failed because the symbol `void tf2::fromMsg(geometry_msgs::Quaternion const&, Eigen::Quaterniond &)` (demangled and simplified) could not be found.
You can reproduce it in a container with geometry2 and moveit installed from source, see below.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04, ros:melodic-ros-base image
* source installed melodic-devel branch, [rosinstall file](https://gist.github.com/gleichdick/152c0b3a7932ee772d4d6342b83b138f)

### Steps to reproduce

Build and run docker container (takes a while tho)

```Dockerfile
FROM ros:melodic-ros-base

RUN apt-get update && apt-get upgrade -y && apt-get install -y python3-pip

# force installation of geometry2 from source
RUN apt-mark hold ros-melodic-tf2-eigen

# install catkin tools from source because .deb package is broken
RUN git clone --depth=1 https://github.com/catkin/catkin_tools /catkin_tools && \
   cd /catkin_tools && pip3 install -e .

WORKDIR /ws_moveit

RUN wstool init src https://gist.github.com/gleichdick/152c0b3a7932ee772d4d6342b83b138f/raw/b63b050d1362cdea3de05573de62f838bfa11241/moveit.rosinstall && \
    wstool update -t src && \
    catkin config --extend /opt/ros/melodic --install --install-space /ws_moveit/install --cmake-args -DCMAKE_BUILD_TYPE=Release

RUN rosdep install --from-paths src --ignore-src --rosdistro melodic -y

RUN catkin build
```

Inside the container:

```sh
$ source /ws_moveit/install/setup.sh
$ python
Python 2.7.15+ (default, Oct  7 2019, 17:39:04) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import moveit_commander
```

### Expected behaviour

Module should be imported

### Actual behaviour

Module could not be imported because symbol

```c++
void tf2::fromMsg<geometry_msgs::Quaternion_<std::allocator<void> >, Eigen::Quaternion<double, 0> >(geometry_msgs::Quaternion_<std::allocator<void> > const&, Eigen::Quaternion<double, 0>&)
```

could not be found which leads to an `ImportError`

### Backtrace or Console output

```sh
$ python
Python 2.7.15+ (default, Oct  7 2019, 17:39:04) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import moveit_commander
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/ws_moveit/install/lib/python2.7/dist-packages/moveit_commander/__init__.py", line 4, in <module>
    from .move_group import *
  File "/ws_moveit/install/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 41, in <module>
    from moveit_ros_planning_interface import _moveit_move_group_interface
ImportError: /ws_moveit/install/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_move_group_interface.so: undefined symbol: _ZN3tf27fromMsgIN13geometry_msgs11Quaternion_ISaIvEEEN5Eigen10QuaternionIdLi0EEEEEvRKT_RT0_
>>>
```

### Additional info

[ldd _moveit_move_group_interface.so](https://gist.github.com/gleichdick/24094cd74296b43e36a846c8d229481d)

```sh
$ objdump -t /ws_moveit/install/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_move_group_interface.so | grep fromMsg
0000000000000000         *UND*  0000000000000000              _ZN3tf27fromMsgIN13geometry_msgs11Quaternion_ISaIvEEEN5Eigen10QuaternionIdLi0EEEEEvRKT_RT0_
```

```sh
$ ls -l /ws_moveit/install/lib/python2.7/dist-packages/moveit_ros_planning_interface/
total 1196
-rw-r--r-- 1 root root      0 Nov 27 14:55 __init__.py
-rw-r--r-- 1 root root    175 Nov 27 16:05 __init__.pyc
lrwxrwxrwx 1 root root     37 Nov 27 16:05 _moveit_move_group_interface.so -> _moveit_move_group_interface.so.1.0.2
-rw-r--r-- 1 root root 686800 Nov 27 16:05 _moveit_move_group_interface.so.1.0.2
lrwxrwxrwx 1 root root     41 Nov 27 16:05 _moveit_planning_scene_interface.so -> _moveit_planning_scene_interface.so.1.0.2
-rw-r--r-- 1 root root 182848 Nov 27 16:03 _moveit_planning_scene_interface.so.1.0.2
lrwxrwxrwx 1 root root     32 Nov 27 16:05 _moveit_robot_interface.so -> _moveit_robot_interface.so.1.0.2
-rw-r--r-- 1 root root 302992 Nov 27 16:03 _moveit_robot_interface.so.1.0.2
lrwxrwxrwx 1 root root     35 Nov 27 16:05 _moveit_roscpp_initializer.so -> _moveit_roscpp_initializer.so.1.0.2
-rw-r--r-- 1 root root  28200 Nov 27 16:03 _moveit_roscpp_initializer.so.1.0.2
```

gleichdick, 2019-11-27:
This might actually be a bug in tf2_eigen.

In `<tf2/transform_functions.h>` (which is included at the top of `<tf2_eigen/tf2_eigen.h>`) the following template is declared:

```c++
template<typename A, typename B>
  void fromMsg(const A&, B& b);
```

Oviously no implementation is provided. In `<tf2_eigen/tf2_eigen.h>`, the missing symbol is defined as
```c++
inline
void fromMsg(const geometry_msgs::Quaternion& msg, Eigen::Quaterniond& out) {
  out = Eigen::Quaterniond(msg.w, msg.x, msg.y, msg.z);
}
```
Note that this is **not** a specialisation of the template above because `template<>` is missing.
So, I added the `template<>` specifier in tf2_eigen.h for each `fromMsg` overload, it compiles and works now.

Should this be discussed with the geometry2 package people? ros/geometry2#111 introduced the template...

rhaschke, 2019-11-28:
Thanks for reporting this issue and providing a detailed analysis. Interestingly, the issue doesn't occur with the currently released packages. So I looked a little bit deeper: Looks like the issue was introduced in https://github.com/ros/geometry2/pull/367. Can you validate this? I don't have the time to do so.
If confirmed, could you please file your detailed issue report over at https://github.com/ros/geometry2?

gleichdick, 2019-11-28:
Yep, reverting ros/geometry2#367 (by checking out the commit before it) fixed it. Thanks for the hint. Issue ros/geometry2#430 opened.