moveit/moveit #1791: Sync cartesian path for dual arms

hi-kondo, 2019-11-29:
I'm using manipulator robot with dual arms. 
I would like to make it move both arms with synchronize for cartesian path.
But I think that it is impossible to implement...
Do you have any ideas?

rhaschke, 2019-11-29:
MoveIt can plan trajectories for any robots, also dual arm ones. For execution MoveIt hands the trajectories to FollowJointControllers and it's their task to execute them according to the given time schedule to stay in sync. So, if you have appropriate controllers that respect the time evolution - go ahead.
Please post such questions on ROS answers. The github issues are reserved for real issues.

closed 2019-11-29 by rhaschke

felixvd, 2019-11-30:
@hi-kondo Please post the link to your ROS Answers post (which should have some more detail on what you're doing and what you have tried) here when you post it.