moveit/moveit #1798: Removing objects can leave relics in rviz

Dale-Koenig, 2019-12-03:
### Description
When using multiple nodes (or restarting a node), removing an object from the planning scene using MoveitVisualTools or PlanningSceneMonitor`s `processCollisionObjectMsg` will leave a green relic in rviz rather than removing the object.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
master branch

### Steps to reproduce
Given a node that brings up MoveitVisualTools and properly connects it to the planning scene

1) Make two copies of the node with the following two code snippets differing between them

Node 1:
```
  /* RViz provides many types of markers, in this demo we will use text, cylinders, and spheres*/
  Eigen::Isometry3d pose = Eigen::Isometry3d::Identity();
  pose.translation().z() = 1.0;
  visual_tools.publishCollisionCuboid(pose, .3, .3, .3, "object",
                                       rvt::ORANGE);
  moveit_msgs::CollisionObject co;
  co.id = "object2";
  co.operation = moveit_msgs::CollisionObject::REMOVE;
  visual_tools.processCollisionObjectMsg(co);
  visual_tools.trigger();
```
Node 2:
```
  Eigen::Isometry3d pose = Eigen::Isometry3d::Identity();
  pose.translation().z() = 1.50;
  visual_tools.publishCollisionCuboid(pose, .3, .3, .3, "object2", rvt::ORANGE);
  moveit_msgs::CollisionObject co;
  co.id = "object";
  co.operation = moveit_msgs::CollisionObject::REMOVE;
  visual_tools.processCollisionObjectMsg(co);
  visual_tools.trigger();

2) Bring up roscore and rviz
3) Run node 1, then shut it down.
4) Run node 2
```
### Expected behaviour
"object" is removed from rviz, and "object2" appears in rviz

### Actual behaviour
"object" remains in rviz, but in `rvt::GREEN`

### Backtrace or Console output
[image]

simonschmeisser, 2019-12-06:
This seems to be a bit of a bug in MoveIt visual tools here: https://github.com/ros-planning/moveit_visual_tools/blob/af72ea8c86ce5900d7f44e695b4db6fa430860c9/src/moveit_visual_tools.cpp#L138
as in the case of DELETE it does not make sense to add a color. Note that for most likely historical reasons colors and objects are stored separately in PlanningScene. Calling setObjectColor for a not existing object might confuse things a bit?

Could you explain a bit more about your setup? Because if you just have a plain node with just an instance of MoveItVisualTools then I would not expect changes to propagate at all?

Ah ok, here is how it seems to work:
both nodes have a unique and completely separate PlanningScene. However they both publish this to the same topic.
https://github.com/ros-planning/moveit_visual_tools/blob/af72ea8c86ce5900d7f44e695b4db6fa430860c9/src/moveit_visual_tools.cpp#L114
So rviz will show a mixture of both states.

Note that if you do path planning in a third node, typically MoveGroup node, this has its own third PlanningScene and will not regard your cubes as obstacles!

Dale-Koenig, 2019-12-06:
Thanks, yeah it seems it is maybe just "unexpected behavior" more than a bug.  I would expect a non-diff msg to be sent once to reset the scene upon creating a new planning scene monitor.  Without doing this, it makes it hard to maintain the visualization when a node is shut down and restarted.  I suppose a hack to solve it would be to publish the object and immediately remove it, to guarantee that it gets removed with a diff message.

The extent of the bug may just be that visual tools sets the color to green (the default argument to `processCollisionObjectMsg`) even when the msg is supposed to be removing the object.