moveit/moveit #1800: Adding pilz_industrial_motion to this repo

davetcoleman, 2019-12-04:
Moving https://github.com/PilzDE/pilz_industrial_motion/issues/228 discussion here.

We have been talking with Pilz, and they have agreed, to further integrate [their industrial planners](https://github.com/PilzDE/pilz_industrial_motion) (generators to create LIN, PTP and CIRC trajectories) into the MoveIt project by merging their planners here.

**Benefits**
- More users / exposure of their great work to the MoveIt community
- Fill a much needed gap in MoveIt's feature set for classic industrial interpolation approaches to planning
- Reduce duplicate maintenance by including code cleanup, releasing, upgrading code, and general community policy into the MoveIt maintainer project
- Increase MoveIt's test coverage (as measured by the [new CodeCov](https://github.com/ros-planning/moveit_ci/pull/91)!) by including Pilz's very feature rich test coverage
- Include more companies (Pilz) in the development of MoveIt 

**TODO**
- [ ] Decide on name for this new planner (``industrial_planner`` was suggested by Pilz in meeting, but they are open to renaming)
- [ ] Add @jspricke as a MoveIt maintainer so he can continue to maintain this feature. I think he has proven his skills with the development of pilz_industrial_motion
- [ ] Add @ct2034 as a MoveIt Core Contributor so he can also get more involved in maintenance
- [ ] Create PR to merge into ``master`` branch
- [ ] Backport PR to ``melodic``
- [ ] Optional: Backport PR to ``kinetic``? In our meeting it was not clear if this is necessary. Personally on further thought, I don't think it's worth it.
- [ ] Create moveit_tutorial for this new planner
- [ ] Create blog post about this collaboration

jspricke, 2019-12-04:
Hey @davetcoleman, thanks for the offer but I guess you don't mean me ;).

gavanderhoorn, 2019-12-04:
Much of the Pilz work is GPLv3, which is not something we'd want to import into MoveIt. Changing licenses is possible, but would require an OK from all contributors.

Additionally: `pilz_industrial_motion` has been supported by the ROSIN project, which leads to a requirement for Pilz to make sure maintenance and access to the results are guaranteed up to a certain nr of years. Has this been discussed as well?

ct2034, 2019-12-04:
@davetcoleman thanks for the beautiful summary! But please change the user handle to @jschleicher.

ct2034, 2019-12-04:
And I want to add
( ) make @jschleicher and @ct2034 code owners

davetcoleman, 2019-12-05:
Sorry @jspricke for tagging the wrong name! Fixed.

davetcoleman, 2019-12-05:
@gavanderhoorn I had not realized its GPLv3, that is very concerning!

davetcoleman, 2020-01-13:
@jschleicher any update on changing the license of pilz_industrial_motion to non-copy left?

jschleicher, 2020-02-06:
@davetcoleman Yep, we'll change the license of the affected `pilz_trajectory_generation` package. Starting to work on that we'll post a PR soon @ct2034

davetcoleman, 2020-12-04:
@jschleicher did this get backported to Melodic as desired?