moveit/moveit #1801: Unable to save states/scenes from RVIZ to mongod database

gokul-gokz, 2019-12-06:
I am trying to save states and scenes from rviz to mongod database to benchmark planners for ur3 arm. But whenever I try to save them, I am getting the following error.

`Cannot save robot state on the database: filemd5 failed
`
1. I have tried in both Ros-kinetic and melodic. I am getting the same issue. I also tried source installation. 
2. I have followed these [steps](https://moveit.ros.org/install/source/dependencies/) to install the mongo driver.
3. I have followed these [steps](https://docs.mongodb.com/manual/tutorial/install-mongodb-on-ubuntu/) to install the mongoDB.
4.I am able to connect to the database from RVIZ but unable to save states.

felixvd, 2020-04-16:
Did you solve this issue?

gokul-gokz, 2020-05-26:
No. I couldn't able to solve @felixvd