moveit/moveit #1805: Segfault in totg

Dale-Koenig, 2019-12-09:
### Description

Segfault in totg, possibly Eigen related

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
Source, master

### Steps to reproduce
Difficult at the moment to reproduce without sharing my code, but likely has to do with having a portion of the plan "static" (for example, where the first 3 steps don't really go anywhere because they come from trying to plan to where you already are)

### Expected behaviour
No segfault

### Actual behaviour
Segfault

### Backtrace or Console output

[backtrace](https://gist.github.com/Dale-Koenig/823d98b3fd033682fe65862c91e8555d)

gavanderhoorn, 2019-12-09:
First suggestion: build things with `Debug` or `RelWithDebInfo` so the trace would show more information.

Dale-Koenig, 2020-01-23:
It seems the (or a) cause of this is when `CircularPathSegment` is constructed with `start` and `end` being opposite directions, so due to floating point errors `start_direction.dot(end_direction) < -1`, and computing `acos` results in `nan`

Dale-Koenig, 2020-01-23:
Related: https://github.com/tobiaskunz/trajectories/issues/4

PR #1861 (2020-02-03): Fix segfault in totg
### Description

(At least partially) fixes  #1805.  Calls to `acos` can result in nan values if not careful due to numerical floating point errors.  If there is a possibility of calling `acos` on values that are approximately `-1`, then it will result in nan, which leads to a segfault later in totg. Also removes an incorrect comment that implies a section would avoid this error, but it actually handles a different case.
closed 2020-02-03 by rhaschke