moveit/moveit #1807: RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)

RDaneelOlivav, 2019-12-10:
### Description
Hi ,
I am connecting to a robot that the connection is slower than usual, and I suspect that
the timeout could be not letting moveit work.
I've seen other issues posted here but none give a solution for this.
This is the most promising one, but doesnt tell how to access this fucntionality through the MoveGroupCommander or any python modules: [https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/action_based_controller_handle.h#L76](url), [https://github.com/ros-planning/moveit/issues/636](ISSUE Closed)
And for what I've seen, it's a recurrent issue.

So how can I make the MoveGroupCommander wait more time?
Where do I have to touch, even if its in the source code of moveit,
to allow me to increase or even make infinite the waiting time?

### Your environment
* ROS Distro: [Kinetic|]
* OS Version: e.g. Ubuntu 16.04
* Binary build fro kinetic ROS

### Steps to reproduce
The best way is to launch this [http://www.rosject.io/l/f1bf66e/](ROSject) and follow the instructions at the bottom.
Just launch the simulation and execute the python command sat the end of the notebook to test that it works. Then pause the simulation and launch the exact same code. You will see that if its paused more than 5 seconds, it gives the error once you unpause the simulation. This is the closest there is to reproduce the same thing that happens in the real robot.

### Expected behaviour
It should move the robot arm

### Actual behaviour
Error is located [https://github.com/ros-planning/moveit/blob/590a9c36d80ab55882d77f4c349c6344680a499b/moveit_commander/src/moveit_commander/move_group.py#L53](File Error)

It gives the error : RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)

rhaschke, 2019-12-11:
According to the PR you referenced, there was introduced a parameter `trajectory_execution/controller_connection_timeout`, which you could add in your moveit_config's `trajectory_execution.launch` file.
https://github.com/ros-planning/moveit/blob/fe39430d971fdfadcec0c7316fa67d41bf48b281/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/action_based_controller_handle.h#L79

guiolpei, 2020-03-23:
I have the same issue.

Ubuntu 16.04 & ROS Kinetic.

Trying the scene in **simulation**, where one of my links is defined by a really heavy STL file. This causes MoveIt to freeze for a while during launch.

> Starting context monitors...
[ INFO] [1584983437.502889873]: Starting scene monitor
[ INFO] [1584983437.507039743]: Listening to '/planning_scene'
[ INFO] [1584983437.507102594]: Starting world geometry monitor
[ INFO] [1584983437.510564988]: Listening to '/collision_object' using message notifier with target frame '/world '
[ INFO] [1584983437.515631657]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1584983442.867871452]: Starting scene monitor
[ INFO] [1584983442.874332134]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1584983442.875764049]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1584983447.875769503]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.18/planning_scene_monitor/src/planning_scene_monitor.cpp:491
[ INFO] [1584983474.744974293]: Constructing new MoveGroup connection for group 'ur10_arm' in namespace ''
[ERROR] [1584983504.764119642]: Unable to connect to move_group action server 'move_group' within allotted time (30s)

The same error appears and there is no planning library available.

I have tried to include in `trajectory_execution.launch.xml` the following line:

> <param name="trajectory_execution/controller_connection_timeout" value="0.0"/>

But it keeps showing the same message informing about the 30s timeout expiration.

v4hn, 2020-03-23:
@guiolpei You're problem is not that *the move_group node* does not wait long enough (this is what you can fix by setting the `controller_connection_timeout parameter).
Your problem is that you additionally start another node (rviz?) together with the move_group node, and the other node times out after 30 seconds.

Either start your other nodes once move_group is ready, or incorporate an additional waiting period in your nodes, checking the move_group interfaces for existence.

guiolpei, 2020-03-23:
I am indeed starting RViz after the move_group node, it is in the MotionPlanning->Context tab where no _Planning Library_ appears.

Who is causing the timeout then? I do not fully understand your comment.

v4hn, 2020-03-23:
If it *is* rviz, you can probably simply deactivate/activate the Motion Planning display once the `move_group` started successfully.

Probably the following happens:

`move_group` starts up, but hangs on processing the meshes (this is a bug itself in your setup - the system should not need to handle such detailed meshes).
`rviz` starts up alongside with the `MotionPlanning` display enabled.
the display instantiates a `PlanningSceneMonitor` and calls `requestPlanningSceneState` on it, timing out after 5 seconds already.

Additionally, the display instantiates a `MoveGroupInterface` with a timeout of 30 seconds.

both would need to be adjusted to have the display wait (asynchronously) for MoveIt.
Nobody changed it so far, as the `move_group` node should really not take that long to startup - and if it does you can always start rviz by hand once your system is running...

guiolpei, 2020-03-24:
Thank you for the answer, I see what is going on now...

I have reduced the models to obtain a lighter STL file to avoid the timeout, I think it is the most logical think to do. Anyways, maybe I have to look into the solution you suggested in the future.

mremson, 2020-06-05:
Hi I have  similar problem. 
I have installed ros-melodic on Ubuntu 18.04 and built the universal_robot package from source as described in the repository https://github.com/ros-industrial/universal_robot. 
I already tried launching the joint limited version of ur5 but I still get the RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (30s). And I assume this causes that there is no planning library loaded. 

I am running the following commands, each in a different command window:

roslaunch ur_gazebo ur5.launch limited:=true

roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true

roslaunch ur5_moveit_config moveit_rviz.launch config:=true

gavanderhoorn, 2020-06-05:
@mremson: your post seems off-topic here. It should probably be posted on the issue tracker of `universal_robot`, seeing as you're using those packages.

gerardcanal, 2020-07-03:
Hello,
I am finding this issue too in melodic. It seems to happen only when using the python interface (in C++ it connects without issues).

In my case, the first MoveGroupCommander of the node is created without issues, while the next one fails with the 5s message. I'm not aware of any way to specify infinite time from Python, is there one? 
I also observed that is I unset the sensor yaml file (so no cloud processing is performed), it is able to connect to all the move groups without issue.

When tried from C++, I was able to connect to all the move_groups of the robot almost instantly.

antbiv, 2020-07-21:
> I have the same issue.
> Ubuntu 16.04 & ROS Kinetic.
> Trying the scene in simulation, where one of my links is defined by a really heavy STL file. This causes MoveIt to freeze for a while during launch.
> 
> Starting context monitors...
> [ INFO] [1584983437.502889873]: Starting scene monitor
> [ INFO] [1584983437.507039743]: Listening to '/planning_scene'
> [ INFO] [1584983437.507102594]: Starting world geometry monitor
> [ INFO] [1584983437.510564988]: Listening to '/collision_object' using message notifier with target frame '/world '
> [ INFO] [1584983437.515631657]: Listening to '/planning_scene_world' for planning scene world geometry
> [ INFO] [1584983442.867871452]: Starting scene monitor
> [ INFO] [1584983442.874332134]: Listening to '/move_group/monitored_planning_scene'
> [ INFO] [1584983442.875764049]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
> [ INFO] [1584983447.875769503]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.18/planning_scene_monitor/src/planning_scene_monitor.cpp:491
> [ INFO] [1584983474.744974293]: Constructing new MoveGroup connection for group 'ur10_arm' in namespace ''
> [ERROR] [1584983504.764119642]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
> 
> The same error appears and there is no planning library available.
> I have tried to include in trajectory_execution.launch.xml the following line:
> 
> param name="trajectory_execution/controller_connection_timeout" value="0.0"
> 
> I have tried with several different values, but it keeps showing the same message informing about the 30s timeout expiration.

Has anybody developed a solution for this. We have the exact same problem with UR5 arms. 

ROS Distro: [Kinetic|
OS Version: e.g. Ubuntu 16.04
Binary build for kinetic ROS

Anybody know of any workarounds?
- We have reduced the robot model to no more than 3Mb
- We have added the wait_for_server logic to MoveIt as described and tried setting it(Need confirmation on how to set)
- We have connected and checked communication of topics in both directions which is fine at 20hz
- Latency is around 250ms

We really need to resolved URGENTLY can anybody assist?

~~~~~~~~~~~~~~~~~~~~~~`
`      Please HELP !!!                      ~
~~~~~~~~~~~~~~~~~~~~~~

thanks
Anthony

felixvd, 2020-07-21:
Volunteer forums aren't the best place to ask for urgent help. It's also unclear to me what exactly you are doing and if you tried the previous suggestions, like starting the `move_group` before RViz.

gachiemchiep, 2020-08-18:
@antbiv 

I'm using the [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) driver.
You have to wait until this line "You can start planning now!" appear in the console. Then you can start planning.

[image]

```

FrancoisTeyssere, 2020-12-03:
@mremson 
@antbiv 

I faced the same problem when trying to run UR5 demo. I was simply missing the package gazebo_ros_control, in case it can help anyone. Not related to the initial issue though.

kyouma9s, 2021-06-23:
I faced the same problem when a python program try to create more than one instance of moveit_group commander.

jiannan0721, 2021-08-11:
> Hi I have similar problem.
> I have installed ros-melodic on Ubuntu 18.04 and built the universal_robot package from source as described in the repository [...]

Hi, we got a very similar issue. Did you resolve this?

Ogunniran, 2023-02-15:
In case you have a script to send the planning commands, after lunching the moveit launch, In your script, put a sleep before initializing the commander.
```
rospy.sleep(10)
move_group_ = moveit_commander.MoveGroupCommander(
                robot_name)
```

yeshwanthguru, 2023-06-19:
[INFO] [1686839270.268691371]: Loading robot model 'tiago_dual'...
[INFO] [1686839270.523448468]: No root/virtual joint specified in SRDF. Assuming fixed joint
[WARN] [1686839303.078767006]: Kinematics solver doesn't support #attempts anymore, but only a timeout. Please remove the parameter '/robot_description_kinematics/arm_left/kinematics_solver_attempts' from your configuration.
Traceback (most recent call last):
File "tiago_grasping.py", line 116, in <module>
move_group = moveit_commander.MoveGroupCommander('arm_right_torso')
File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 66, in init
name, robot_description, ns, wait_for_servers
RuntimeError: Unable to connect to move_group action server 'pickup' within the allotted time (5s)
I have the same issue the script works fine in my simulation environment but is unable to run in a real robot it gets the following error any suggestion with respect to this? Thank you in advance....