moveit/moveit #1810: MoveIt! Perception Pipeline Tutorial fail to construct the octomap

lcmai, 2019-12-14:
Hi, I am following the MoveIt! Perception Pipeline Tutorial and when I try to run 

`roslaunch moveit_tutorials obstacle_avoidance_demo.launch`

I get nothing but the result same as running

`roslaunch panda_moveit_config demo.launch`

and the console outputs are as follow:
~~~
SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'joints': ['pan...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/initial: [{'pose': 'ready'...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
 * /move_group/planner_configs/FMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/FMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/FMTkConfigDefault/radius_multiplier: 1.1
 * /move_group/planner_configs/FMTkConfigDefault/type: geometric::FMT
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
[... 210 lines cut ...]
[ INFO] [1576302969.812405815]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1576302969.907185968]: Stereo is NOT SUPPORTED
[ INFO] [1576302969.907303384]: OpenGl version: 3 (GLSL 1.3).
0x1299340 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x6f0880) ): Attempt to set a screen on a child window.
0x128f320 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x6f0880) ): Attempt to set a screen on a child window.
0x128fbb0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x6f0880) ): Attempt to set a screen on a child window.
0x1295a70 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x6f0880) ): Attempt to set a screen on a child window.
[ INFO] [1576302970.259530224]: Loading robot model 'panda'...
[ INFO] [1576302970.803439222]: Loading robot model 'panda'...
[ INFO] [1576302970.879332389]: Loading robot model 'panda'...
[ INFO] [1576302970.951475491]: Starting scene monitor
[ INFO] [1576302970.956275741]: Listening to '/move_group/monitored_planning_scene'
[ WARN] [1576302970.958025661]: No transform available between frame 'temp_link' and planning frame '/world' (Could not find a connection between 'world' and 'temp_link' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1576302970.958064864]: No transform available between frame 'camera_rgb_optical_frame' and planning frame '/world' (Could not find a connection between 'world' and 'camera_rgb_optical_frame' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1576303019.327252677]: MessageFilter [target=/world ]: Dropped 100.00% of messages so far. Please turn the [ros.moveit_ros_perception.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1576303079.327679865]: MessageFilter [target=/world ]: Dropped 100.00% of messages so far. Please turn the [ros.moveit_ros_perception.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1576303139.327909282]: MessageFilter [target=/world ]: Dropped 100.00% of messages so far. Please turn the [ros.moveit_ros_perception.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1576303199.328182748]: MessageFilter [target=/world ]: Dropped 100.00% of messages so far. Please turn the [ros.moveit_ros_perception.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1576303259.338625536]: MessageFilter [target=/world ]: Dropped 100.00% of messages so far. Please turn the [ros.moveit_ros_perception.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1576303319.348346248]: MessageFilter [target=/world ]: Dropped 100.00% of messages so far. Please turn the [ros.moveit_ros_perception.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1576303379.348482839]: MessageFilter [target=/world ]: Dropped 100.00% of messages so far. Please turn the [ros.moveit_ros_perception.message_notifier] rosconsole logger to DEBUG for more information.
~~~

askkvn, 2020-04-28:
Duplicate of #1220

rhaschke, 2020-04-28:
@mlautman, can you have a look at this in #1220, please?

closed 2020-04-28 by rhaschke