moveit/moveit #1819: ImportError: cannot import name MoveGroupCommander?

Musyue, 2019-12-27:
### Description

Hi, guys,
When I use moveit to simulate our robot, It was ok at first.
But when I compiling OMPL from source code and make install, then I go back to use ros moveit to control our robot. It throws an error. like this:
Traceback (most recent call last):
  File "/data/ros/code/ecto_1_ws/src/ros_exploring/robot_marm/marm_planning/scripts/moveit_obstacles_demo.py", line 6, in <module>
    from moveit_commander import MoveGroupCommander, PlanningSceneInterface
ImportError: cannot import name MoveGroupCommander

I try to uninstall the ros-melodic-moveit-commander, and reinstall, however, it still had the error.
I have no idea about this, and I guess it may be because I install the OMPL.
could you help me?
Thank you
### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Moveit version, I just reinstall today

rhaschke, 2019-12-30:
The python error is probably unrelated to your OMPL source build. I guess you screwed your ROS workspace somehow. Did you correctly define your underlay? Did you correctly source your new workspace? Have a look at your environment, particularly `ROS_PACKAGE_PATH`, `PYTHONPATH`, and `LD_LIBRARY_PATH`. They all should list your local workspace pathes before `/opt/ros` pathes.

Musyue, 2019-12-30:
@rhaschke Thanks for your reply. I solved it by deleting the workspace's build and devel folder. And Everything is fine. Thank you very much. I will close this bug. Sorry to bother you.

closed 2019-12-30 by Musyue

thebirdgr, 2023-03-19:
yes deleting the build and devel folders and rebuilding the workspace with catkin_make helped.