moveit/moveit #1820: Possible memory leak in MoveIt's usage of geometric_shapes

mlautman, 2019-12-30:
### Description

I have noticed a potential memory leak in MoveIt's usage of `geometric_shapes` for the `planning_scene`. After adding and removing boxes thousands of times to different `planning_scene::PlanningScenePtr`s, removing boxes starts to take an incredibly long time. This can be demonstrated by modifying [`suction_grasp_pipeline_demo.cpp`](https://github.com/ros-planning/moveit_grasps/blob/master/src/demo/suction_grasp_pipeline_demo.cpp) to filter grasps with the `moveit_grasps::SuctionGraspFilter` > 10 times in a single run. (see: `suction_grasp_pipeline_demo.launch`)

I was wondering, what reason do we have for using raw pointers with explicit memory management for the planning scene world and the underlying geometric shapes?

Perhaps @davetcoleman, @rhaschke or @v4hn could weigh in here

*Issue moved from https://github.com/ros-planning/geometric_shapes/issues/123*

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source build of the master branch

mlautman, 2019-12-30:
Some relevant output from valgrind:
```
==21283== 22,464 (11,856 direct, 10,608 indirect) bytes in 39 blocks are definitely lost in loss record 477 of 478
==21283==    at 0x4C3017F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==21283==    by 0x6C7FF87: moveit::core::RobotState::attachBody(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::shared_ptr<shapes::Shape const>, std::allocator<std::shared_ptr<shapes::Shape const> > > const&, std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > > const&, std::set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, trajectory_msgs::JointTrajectory_<std::allocator<void> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, Eigen::Transform<double, 3, 1, 0>, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Eigen::aligned_allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, Eigen::Transform<double, 3, 1, 0> > > > const&) (robot_state.cpp:895)
==21283==    by 0x6C7B912: moveit::core::RobotState::copyFrom(moveit::core::RobotState const&) (robot_state.cpp:171)
==21283==    by 0x6C7B1E3: moveit::core::RobotState::RobotState(moveit::core::RobotState const&) (robot_state.cpp:72)
==21283==    by 0x4FA935E: moveit_grasps::GraspFilter::findIKSolution(std::vector<double, std::allocator<double> >&, std::shared_ptr<moveit_grasps::IkThreadStruct> const&, std::shared_ptr<moveit_grasps::GraspCandidate> const&) const (grasp_filter.cpp:566)
==21283==    by 0x4FA4492: moveit_grasps::GraspFilter::filterGraspByGraspIK(std::shared_ptr<moveit_grasps::GraspCandidate> const&, std::vector<double, std::allocator<double> >&, std::shared_ptr<moveit_grasps::IkThreadStruct> const&) const (grasp_filter.cpp:253)
==21283==    by 0x4FA911C: moveit_grasps::GraspFilter::processCandidateGrasp(std::shared_ptr<moveit_grasps::IkThreadStruct> const&) (grasp_filter.cpp:539)
==21283==    by 0x4FFD432: moveit_grasps::SuctionGraspFilter::processCandidateGrasp(std::shared_ptr<moveit_grasps::IkThreadStruct> const&) (suction_grasp_filter.cpp:154)
==21283==    by 0x4FAE26B: moveit_grasps::GraspFilter::filterGraspsHelper(std::vector<std::shared_ptr<moveit_grasps::GraspCandidate>, std::allocator<std::shared_ptr<moveit_grasps::GraspCandidate> > >&, std::shared_ptr<planning_scene::PlanningScene> const&, moveit::core::JointModelGroup const*, std::shared_ptr<moveit::core::RobotState> const&, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [clone ._omp_fn.0] (grasp_filter.cpp:451)
==21283==    by 0x9B0A95D: ??? (in /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0)
==21283==    by 0xADD56DA: start_thread (pthread_create.c:463)
==21283==    by 0x8C7888E: clone (clone.S:95)
```

davetcoleman, 2019-12-31:
I'm not convinced refactoring this area to use shared_ptrs is worth it, since it was probably designed with raw pointer for faster collision checking. I'd say just keep tracking down where that memory is suppose to  be freed but isn't. Maybe @j-petit has some insights?

mlautman, 2019-12-31:
I tracked down the issue to my usage of attachBody. It seems existing attached objects were being overwritten by new ones with the same ID's rather than being deleted.

closed 2019-12-31 by mlautman