moveit/moveit #1824: Self collision check fails

shyamashi, 2020-01-03:
I have created a URDF file and used MoveIt Setup Assistant to create a package. Although I generated the collision matrix using MSA, when I do simulations in Rviz, the robot collides with itself. 
[image]

I am using ROS melodic, Ubuntu 18.04. All the files required are available in my [repo](https://github.com/shyamashi/free_floating_7_dof_arm). 
Steps to reproduce:
``` roslaunch free_floating_7_dof_arm demo.launch ```
Move the robot using interactive marker and you can see it going inside the base square.

Thanks for any help in advance

rhaschke, 2020-01-04:
I think you misconceived the semantics of the allowed collision matrix (ACM):
If a link pair is ticked that means that this pair _is allowed_ to collide / doesn't need collision checking.
According to your ACM: most collisions are explicitly enabled:
[image]

Hugal31, 2024-09-25:
I find this confusing: [moveit_msgs/AllowedCollisionEntry](https://github.com/moveit/moveit_msgs/blob/master/msg/AllowedCollisionEntry.msg) says
> \# whether or not collision ***checking*** is enabled
> bool[] enabled

IMO

rhaschke, 2024-09-25:
I agree that the wording in the comment is misleading.