moveit/moveit #1827: meta package dependencies and topological order

S-Fichtl, 2020-01-07:
### Description
Running `catkin_topological_order` returns packages in an unbuildable order for moveit.
Specifically, it lists some meta packages earlier than it should.
The reason for this is that the meta package package.xml files only have run dependencies listed, but not build dependencies, making catkin think building those can happen at any time, which is not the case. Building a meta package without having built the actual content packages first will fail.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? Source
* If source, which branch? 1.0.2

### Steps to reproduce
in a working moveit workspace, run `catkin_topological_order`

### Expected behaviour
a list should be generated listing all packages in an order that reflects their build dependency order, such that if you build them one by one in the order as suggested by the above command, all packages should be buildable.
Specifically, the list should have the meta packages (`moveit`, `moveit_runtime`, `moveit_ros`, `moveit_plugins` and `moveit_planner`) *after* the actual content packages (e.g. `moveit` and `moveit_runtime` should be at the bottom of the list, not at the top)

### Actual behaviour
the generated list starts with `moveit` and also lists the other meta packages earlier than it should, causing scripts that build packages following the order suggested by `catkin_topological_order` to fail because a meta package can't be built before the content packages are ready.

### Backtrace or Console output
No output here, but a suggestion for a fix: use build_depend in package.xml of meta packages instead of exec_depend.

gavanderhoorn, 2020-01-07:
Afaik, metapackages are not allowed to contain anything but `exec_depend`s.

Metapackages are also not meant to be "build". In a typical `catkin_tools` build they are also skipped.

From [REP-140: Package Manifest Format Two Specification - Data Representation - \<export> - \<metapackage/>](https://www.ros.org/reps/rep-0140.html#metapackage):

> This empty tag declares a special kind of catkin package used for grouping other packages. Metapackages only provide execution-time dependencies. They cannot be used for catkin builds and compile nothing themselves. Metapackages may not install any code or other files, although `package.xml` does get installed automatically. They can depend on other metapackages, but regular catkin packages cannot.

And:

> Because metapackages only supply execution-time dependencies, they use `<exec_depend>` to list the packages in their group: [..]

So I don't believe this is something that would need any fixing here in MoveIt.

Perhaps ask this over at the Catkin repository.

gavanderhoorn, 2020-01-07:
I'm inclined to close this as a `works-for-me`, but I'll wait for your response.

S-Fichtl, 2020-01-07:
I see. I wonder why the catkin folks set this up as exec-time only dependencies, but seems indeed expected behaviour and not at all an issue in moveit.

closed 2020-01-07 by S-Fichtl

gavanderhoorn, 2020-01-07:
> I wonder why the catkin folks set this up as exec-time only dependencies

Just a note: package manifests are not a "Catkin-thing". The spec for that is completely Catkin independent.

I only referred you to "the Catkin repository" as you mentioned using `catkin_topological_order`. It may be that tool needs a fix/update, but that would be something to report over there.

I doubt anything'll change though: Catkin has been around for a long time, and I'm not sure why this wouldn't have been an issue before.