moveit/moveit #1835: Issues with PlanningScene updates

rhaschke, 2020-01-16:
As pointed out in https://github.com/ros-planning/moveit_task_constructor/issues/122#issuecomment-574659080, PlanningScene updates [performed by MTC's `ExecuteTaskSolutionCapability`](https://github.com/ros-planning/moveit_task_constructor/blob/ef6cb1a746353bf2b11eb986c9f690e1c785e623/capabilities/src/execute_task_solution_capability.cpp#L179), are broken in the master branch (latest binary release still works!).

If I augment the mentioned call with a 0.5s sleep, the spurious collision is gone and trajectory execution finishes, but the PlanningScene still isn't correctly updated:

|melodic: ok  |master: broken|
|---|---|
|[image]|[image] |

Obviously, the object is not correctly unattached. Interestingly, the same process (but maybe via a different mechanism) works during planning. Probably there is a race condition as well because adding a pause partially fixed the issue.

felixvd, 2020-01-16:
I'll double check tomorrow if #1439 is involved, since that involved changes in the `processAttachedCollisionObject` and `processCollisionObject` methods. For some reason I can't run the MTC demo at all at the moment - Rviz does not show a robot or the example task.

rhaschke, 2020-01-16:
#1439 is not the culprit. I'm currently bisecting the issue. Currently, #1808 is still good.

rhaschke, 2020-01-16:
#1439 is not the culprit. Bisecting the issue shows that #1584 is the culprit.
@j-petit, do you have any idea, why the unified collision environment might exhibit a race condition?
I remember that we had a similar issue in the past.

j-petit, 2020-01-17:
Hm, not off the top of my head... Is this is executed as multi-threaded collision checking?  I'll try to look more closely at the code again today.

>  I remember that we had a similar issue in the past.

Before the introduction of the unified environment or after?

rhaschke, 2020-01-17:
> I remember that we had a similar issue in the past.

After introducing the unified environment.

The issue occurs for a single-thread environment using FCL.

j-petit, 2020-01-17:
What might be the reason is that [the callback](https://github.com/ros-planning/moveit/blob/06a9e4beac1e94d862c8286a22561529cb03d638/moveit_core/collision_detection/src/world.cpp#L319) which deletes an object when it changes from environment to being attached is executed only after the object is added to the `robot_state` as being attached and a collision check performed.

rhaschke, 2020-01-17:
The PlanningScene's `World` and `RobotState` is correct before the collision check is ran as the following debug output illustrates (_attached_ and _free_ object lists correctly list the right objects):
```
[DEBUG] ros.moveit_task_constructor_capabilities.ExecuteTaskSolution: apply effect of 8/17
attached: 0x7f14b4001770 object, 
free: 0x7f14d400ccb0 table, 
[ INFO] ros.moveit_core.collision_detection.fcl: Found a contact between 'object' (type 'Object') and 'object' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
```
This is the stack trace:
```
  libmoveit_collision_detection_fcl.so.1.0.1 : collision_detection::collisionCallback(fcl::CollisionObject*, fcl::CollisionObject*, void*)+0xfff
  /usr/lib/x86_64-linux-gnu/libfcl.so.0.5 : fcl::details::dynamic_AABB_tree::collisionRecurse(fcl::NodeBase<fcl::AABB>*, fcl::CollisionObject*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*))+0xc3
  /usr/lib/x86_64-linux-gnu/libfcl.so.0.5 : fcl::DynamicAABBTreeCollisionManager::collide(fcl::CollisionObject*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*)) const+0x142
  libmoveit_collision_detection_fcl.so.1.0.1 : collision_detection::CollisionEnvFCL::checkRobotCollisionHelper(collision_detection::CollisionRequest const&, collision_detection::CollisionResult&, moveit::core::RobotState const&, collision_detection::AllowedCollisionMatrix const*) const+0x10c
  libmoveit_planning_scene.so.1.0.1 : planning_scene::PlanningScene::checkCollisionUnpadded(collision_detection::CollisionRequest const&, collision_detection::CollisionResult&, moveit::core::RobotState const&, collision_detection::AllowedCollisionMatrix const&) const+0x4f
  libmoveit_plan_execution.so.1.0.1 : plan_execution::PlanExecution::isRemainingPathValid(plan_execution::ExecutableMotionPlan const&, std::pair<int, int> const&)+0x445
  libmoveit_plan_execution.so.1.0.1 : plan_execution::PlanExecution::successfulTrajectorySegmentExecution(plan_execution::ExecutableMotionPlan const*, unsigned long)+0x176
  libmoveit_trajectory_execution_manager.so.1.0.1 : trajectory_execution_manager::TrajectoryExecutionManager::executeThread(boost::function<void (moveit_controller_manager::ExecutionStatus const&)> const&, boost::function<void (unsigned long)> const&, bool)+0x10c
  /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 : ()+0x11bcd
  /lib/x86_64-linux-gnu/libpthread.so.0 : ()+0x76db
  /lib/x86_64-linux-gnu/libc.so.6 : clone()+0x3f
```
As the collision check is performed in the TrajectoryExecutionManager's thread, there might be other threads accessing the collision data structures in parallel.

rhaschke, 2020-01-17:
Indeed the function notifyObserverAllObjects isn't called so often. Maybe there is such a call missing?

rhaschke, 2020-01-18:
I think there is a conceptual problem with the _thread-local_ `FCLShapeCache`:
The thread-local cache is cleaned only in a single thread, although multiple threads might access the same PlanningScene / World. I think, the cache(s) should only get marked invalid (e.g. using a counter in CollisionEnvFCL which then gets larger than an corresponding number in the thread-local cache(s)).
When later accessing the `World` to build/update the cache(s), the `World` needs to be write-protected to avoid race conditions. Before this wasn't an issue, because the cache wasn't thread-local and thus was directly updated for all threads when `cleanCollisionGeometryCache()` was called, which meant that the caller held a `PlanningScene` lock.

Another question I have is: Why are _both_ caches invalidated in `cleanCollisionGeometryCache()`:
https://github.com/ros-planning/moveit/blob/06a9e4beac1e94d862c8286a22561529cb03d638/moveit_core/collision_detection_fcl/src/collision_common.cpp#L907-L917

All 3 call locations handle World updates only. Thus, I think it suffices to invalidate the `World::Object` cache.

Unfortunately, all these issues don't explain the actual issue #1835: Debugging shows that the same thread, namely TrajectoryExecutionManager's executeThread, is manipulating the world, invalidating the cache, and performing the collision check. I also validated that the notify callback is actually called. Hence https://github.com/ros-planning/moveit/issues/1835#issuecomment-575531215 is also excluded.
However, there are other threads, most notably PlanningSceneMonitor's scenePublishingThread, which frequently pushes scene updates to the scene's parent:
https://github.com/ros-planning/moveit/blob/06a9e4beac1e94d862c8286a22561529cb03d638/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L381-L382

j-petit, 2020-01-20:
> Another question I have is: Why are _both_ caches invalidated in `cleanCollisionGeometryCache()`: [...]

Yes, I think that is reasonable.

simonschmeisser, 2020-01-23:
Objects being in the cache should not create collisions by themselves but at worst waste memory? There is no "object persistence" in the fcl plugin but MoveIt actually does all the pairwise checks.

rhaschke, 2020-01-23:
@simonschmeisser, I don't understand your comment. The FCLShapeCache is a data structure of MoveIt, storing all FCL geometries. If there is a geometry represented in both caches (obstacles and attached bodies) this will result in a collision. What do you mean by "object persistence"?

simonschmeisser, 2020-01-23:
What I'm saying is that there is no "World" datastructure in FCL and that FCLShape is not an entity in such a world. FCLShape is just a geometrical description (pre processed mesh, BHV) but not a object at a specific position. One would expect that such a datastructure (ie scene graph) exists in FCL (and it could) but this is not how MoveIt is using FCL. MoveIt just iterates over all pairs of objects and, looks up the corresponding FCLShape using the MoveIt ShapePtr as index and then places one instance of each FCLShape at the pose coming from MoveIt. Then those two FCLShapes are checked for collision. *

Having the same "MoveIt ShapePtr" in both caches is perfectly valid (e.g. we do bin-picking and as such have 50 or more objects with the same MoveIt ShapePtr and then check if lifting one up collides with all the others)

* this is one of the reasons for FCL being "slow" the way MoveIt uses it. I think the bullet plugin has a proper representation of the world/robot.

rhaschke, 2020-01-24:
I think you are missing my point. The problem with the present issue is that the same geometric shape (no matter it is represented by the same ShapePtr or not) is present twice _at the same pose_, which obviously causes a collision.
I agree that it is perfectly fine to share ShapePtrs across multiple object instances. However, in these cases, you use different (hopefully not-colliding) poses.
If you see the potential for improvement in MoveIt's collision handling, don't hesitate to file a PR :wink:

lucarinelli, 2020-01-29:
+1
I think this problem is affecting also the pick pipeline using MoveIt master. https://answers.ros.org/question/342807/moveit-pick-contact-between-object-and-robot-attached-object/

j-petit, 2020-03-01:
Unfortunately, after several hours I am still not very far in debugging the issue... The `DEBUG` statements below indicate where things go wrong:

##### With bug
```
[ INFO] [1583078135.737342975]: Fake execution of trajectory
[DEBUG] [1583078136.237679579]: Adding planning scene diff
[DEBUG] [1583078136.237945963]: Adding planning scene diff
[DEBUG] [1583078136.238118569]: Action 2object: object
[DEBUG] [1583078136.238261248]: Removing world object with the same name as newly attached object: 'object'
[DEBUG] [1583078136.238353836]: Attached object 'object' to link 'panda_link8'
[DEBUG] [1583078136.238460133]: Adding planning scene diff
[DEBUG] [1583078136.238653858]: Checking World collision with 2
[DEBUG] [1583078136.238832040]: Checking World collision with 2
[DEBUG] [1583078136.238930751]: Collision between 'object' (type 'Object') and 'panda_hand' (type 'Robot link') is always allowed. No contacts are computed.
[DEBUG] [1583078136.238973937]: Collision between 'object' (type 'Object') and 'panda_leftfinger' (type 'Robot link') is always allowed. No contacts are computed.
[DEBUG] [1583078136.239011273]: Collision between 'object' (type 'Object') and 'panda_rightfinger' (type 'Robot link') is always allowed. No contacts are computed.
[DEBUG] [1583078136.239067113]: Actually checking collisions between object and object
[ INFO] [1583078136.239128025]: Found a contact between 'object' (type 'Object') and 'object' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
[ INFO] [1583078136.239189256]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1583078136.239260904]: Upcoming trajectory component '10/19' is invalid
[ INFO] [1583078136.239925233]: Fake execution of trajectory
[DEBUG] [1583078136.246649568]: Checking World collision with 2
[DEBUG] [1583078136.246839370]: Checking World collision with 2
[DEBUG] [1583078136.246934697]: Collision between 'object' (type 'Object') and 'panda_hand' (type 'Robot link') is always allowed. No contacts are computed.
[DEBUG] [1583078136.247003580]: Collision between 'object' (type 'Object') and 'panda_leftfinger' (type 'Robot link') is always allowed. No contacts are computed.
[DEBUG] [1583078136.247069780]: Collision between 'object' (type 'Object') and 'panda_rightfinger' (type 'Robot link') is always allowed. No contacts are computed.
[DEBUG] [1583078136.247123170]: Actually checking collisions between object and object
[ INFO] [1583078136.247177314]: Found a contact between 'object' (type 'Object') and 'object' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
[ INFO] [1583078136.247241956]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1583078136.247367796]: Trajectory component '10/19' is invalid after scene update
[ INFO] [1583078136.247424289]: Stopping execution because the path to execute became invalid(probably the environment changed)
[ INFO] [1583078136.340293273]: Fake trajectory execution cancelled
[ INFO] [1583078136.340435301]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1583078136.340680790]: Stopped trajectory execution.
[DEBUG] [1583078136.474822468]: Action 2object: object
[DEBUG] [1583078136.474955608]: Action 2object: object
```

##### With workaround patch
```
[ INFO] [1583078024.703088861]: Fake execution of trajectory
[DEBUG] [1583078025.203489930]: Adding planning scene diff
[DEBUG] [1583078025.303955585]: Adding planning scene diff
[DEBUG] [1583078025.304164963]: Action 2object: object
[DEBUG] [1583078025.304313163]: Removing world object with the same name as newly attached object: 'object'
[DEBUG] [1583078025.304409522]: Attached object 'object' to link 'panda_link8'
[DEBUG] [1583078025.404653426]: Adding planning scene diff
[DEBUG] [1583078025.480423413]: Action 2object: object
[DEBUG] [1583078025.480539576]: Action 2object: object
[DEBUG] [1583078025.505042205]: Checking World collision with 1
[DEBUG] [1583078025.505355627]: Checking World collision with 1
[DEBUG] [1583078025.505562937]: Checking World collision with 1
[DEBUG] [1583078025.505741456]: Checking World collision with 1
[DEBUG] [1583078025.505926072]: Checking World collision with 1
[DEBUG] [1583078025.506294756]: Checking World collision with 1
[DEBUG] [1583078025.506475014]: Checking World collision with 1
[DEBUG] [1583078025.506635295]: Checking World collision with 1
[DEBUG] [1583078025.506799405]: Checking World collision with 1
[DEBUG] [1583078025.506962422]: Checking World collision with 1
[DEBUG] [1583078025.507140891]: Checking World collision with 1
[DEBUG] [1583078025.507309141]: Checking World collision with 1
```

`[DEBUG] [1583078025.480539576]: Action 2object: object` corresponds to removing a collision object from the FCL collision manager. This action is executed three times in the working version, before any checks are done. In the non-patched version, this action is also performed three times, however, two of them only after the collision checking has already started. Consequently, a collision is detected. The number after `Checking World collision with _` indicates how many objects are currently in the FCL collision manager.

Anyone has an idea why we all the actions are repeated three times? Both objects are added/removed three times, `Adding planning scene diff` comes up three times, etc. Is this because of multi-threaded planning?

j-petit, 2020-03-01:
Some backtraces 
##### Backtrace first call to notifyObjectChange callback with `DELETE` as action
Triggered by `trajectory_execution_manager`
```
#0  collision_detection::CollisionEnvFCL::notifyObjectChange (this=..., obj=..., action=...) at /home/petit/ws_moveit/src/moveit/moveit_core/collision_detection_fcl/src/collision_env_fcl.cpp:393
#1  0x00007fffeda823e2 in boost::_mfi::mf2<void, collision_detection::CollisionEnvFCL, std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action>::operator() (
    this=..., p=..., a1=..., a2=...) at /usr/include/boost/bind/mem_fn_template.hpp:280
#2  0x00007fffeda80030 in boost::_bi::list3<boost::_bi::value<collision_detection::CollisionEnvFCL*>, boost::arg<1>, boost::arg<2> >::operator()<boost::_mfi::mf2<void, collision_detection::CollisionEnvFCL, std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action>, boost::_bi::rrlist2<std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action> > (this=..., f=..., a=...) at /usr/include/boost/bind/bind.hpp:398
#3  0x00007fffeda7d66f in boost::_bi::bind_t<void, boost::_mfi::mf2<void, collision_detection::CollisionEnvFCL, std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action>, boost::_bi::list3<boost::_bi::value<collision_detection::CollisionEnvFCL*>, boost::arg<1>, boost::arg<2> > >::operator()<std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action> (this=..., a1=..., a2=...) at /usr/include/boost/bind/bind.hpp:1318
#4  0x00007fffeda7a9d6 in boost::detail::function::void_function_obj_invoker2<boost::_bi::bind_t<void, boost::_mfi::mf2<void, collision_detection::CollisionEnvFCL, std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action>, boost::_bi::list3<boost::_bi::value<collision_detection::CollisionEnvFCL*>, boost::arg<1>, boost::arg<2> > >, void, std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action>::invoke (function_obj_ptr=..., a0=..., a1=...) at /usr/include/boost/function/function_template.hpp:159
#5  0x00007ffff2616fd6 in boost::function2<void, std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action>::operator() (this=..., a0=..., a1=...)
    at /usr/include/boost/function/function_template.hpp:759
#6  0x00007ffff2614d42 in collision_detection::World::notify (this=..., obj=..., action=...) at /home/petit/ws_moveit/src/moveit/moveit_core/collision_detection/src/world.cpp:316
#7  0x00007ffff2614872 in collision_detection::World::removeObject (this=..., object_id=...) at /home/petit/ws_moveit/src/moveit/moveit_core/collision_detection/src/world.cpp:263
#8  0x00007ffff2235b52 in planning_scene::PlanningScene::processAttachedCollisionObjectMsg (this=..., object=...) at /home/petit/ws_moveit/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:1554
#9  0x00007ffff2231455 in planning_scene::PlanningScene::setCurrentState (this=..., state=...) at /home/petit/ws_moveit/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:1109
#10 0x00007ffff22322f9 in planning_scene::PlanningScene::setPlanningSceneDiffMsg (this=..., scene_msg=...) at /home/petit/ws_moveit/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:1207
#11 0x00007ffff2233002 in planning_scene::PlanningScene::usePlanningSceneMsg (this=..., scene_msg=...) at /home/petit/ws_moveit/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:1277
#12 0x00007ffff77f3ef4 in planning_scene_monitor::PlanningSceneMonitor::newPlanningSceneMessage (this=..., scene=...)
    at /home/petit/ws_moveit/src/moveit/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:555
#13 0x00007fffb89bce35 in move_group::ExecuteTaskSolutionCapability::constructMotionPlan(moveit_task_constructor_msgs::Solution_<std::allocator<void> > const&, plan_execution::ExecutableMotionPlan&)::{lambda(plan_execution::ExecutableMotionPlan const*)#1}::operator()(plan_execution::ExecutableMotionPlan const*) const [clone .isra.300] () from /home/petit/ws_moveit/devel/lib/libmoveit_task_constructor_capabilities.so
#14 0x00007ffff48a49df in boost::function1<bool, plan_execution::ExecutableMotionPlan const*>::operator() (this=..., a0=...) at /usr/include/boost/function/function_template.hpp:760
#15 0x00007ffff489e90c in plan_execution::PlanExecution::successfulTrajectorySegmentExecution (this=..., plan=..., index=...)
    at /home/petit/ws_moveit/src/moveit/moveit_ros/planning/plan_execution/src/plan_execution.cpp:499
#16 0x00007ffff48ad10e in boost::_mfi::mf2<void, plan_execution::PlanExecution, plan_execution::ExecutableMotionPlan const*, unsigned long>::operator() (this=..., p=..., a1=..., a2=...)
    at /usr/include/boost/bind/mem_fn_template.hpp:280
#17 0x00007ffff48ac040 in boost::_bi::list3<boost::_bi::value<plan_execution::PlanExecution*>, boost::_bi::value<plan_execution::ExecutableMotionPlan*>, boost::arg<1> >::operator()<boost::_mfi::mf2<void, plan_execution::PlanExecution, plan_execution::ExecutableMotionPlan const*, unsigned long>, boost::_bi::rrlist1<unsigned long> > (this=..., f=..., a=...) at /usr/include/boost/bind/bind.hpp:398
[... cut ...]
#7  0x00007ffff2614872 in collision_detection::World::removeObject (this=..., object_id=...) at /home/petit/ws_moveit/src/moveit/moveit_core/collision_detection/src/world.cpp:263
#8  0x00007ffff222c61b in planning_scene::PlanningScene::pushDiffs (this=..., scene=...) at /home/petit/ws_moveit/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:456
#9  0x00007ffff77f2360 in planning_scene_monitor::PlanningSceneMonitor::scenePublishingThread (this=...)
    at /home/petit/ws_moveit/src/moveit/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:381
```

rhaschke, 2020-03-09:
Fixed via #1899.

closed 2020-03-09 by rhaschke