moveit/moveit #1850: moveit_commander - ImportError undefined symbol _ZN5boost6python23throw_error_already_setEv

lucarinelli, 2020-01-22:
### Description

Trying to use moveit_commander/move group python interface in a python script very similar to the one in the MoveIt tutorials I obtain

`ImportError: /home/lrinelli/ws_moveit/install/lib/libmoveit_py_bindings_tools.so.1.0.1: undefined symbol: _ZN5boost6python23throw_error_already_setEv`

I think it might by related to eigenpy but I'm not sure.

### Your environment
* ROS Distro: Melodic
* OS Version:  Ubuntu 19.10
* Source build of ROS Melodic and MoveIt master
* Python 2.7
* Boost 1.67.0

### Steps to reproduce

* I installed ROS Melodic from source
* Then in a workspace named `ws_moveit` I installed MoveIt from source following the instruction from the tutorial
* The build did not work because of dependencies, I fixed all the dependencies installing packages with apt when possible or adding sources to `ws_moveit/src`, specifically I took eigenpy from https://github.com/stack-of-tasks/eigenpy
* `catkin build` still does not complete successfully because move group pyhton interfaces does not find `eigenpy/eigenpy.hpp`
* I figured out that eigenpy would not be found unless I use `catkin config --install` and then `catkin build`. In this way the build completes without errors.
* When I try to run the python script that uses moveit_commander from another workspace containing the packages to simulate the robotic arm, built on top of `ws_moveit`, I obtain the error reported below.

### Expected behaviour

The script starts moving the robotic arm and printing the goal state for each pose.

### Actual behaviour

I get the error reported below.

### Backtrace or Console output

```
lrinelli@SAN22389:~/edo_ws/src/test_pkg/src$ ./test_print_state.py __ns:=edo
Traceback (most recent call last):
  File "./test_print_state.py", line 6, in <module>
    import moveit_commander
  File "/home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_commander/__init__.py", line 2, in <module>
    from .roscpp_initializer import *
  File "/home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_commander/roscpp_initializer.py", line 35, in <module>
    from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: /home/lrinelli/ws_moveit/install/lib/libmoveit_py_bindings_tools.so.1.0.1: undefined symbol: _ZN5boost6python23throw_error_already_setEv
```

gavanderhoorn, 2020-01-22:
Note: was already posted on ROS Answers ([eigenpy moveit_commander - ImportError undefined symbol](https://answers.ros.org/question/341777)), but I suggested posting it here for better visibility.

rhaschke, 2020-01-22:
Do the unit tests (e.g. `moveit_commander/test/python_moveit_commander.test`) actually work? 
The demangled symbol name is: `boost::python::throw_error_already_set()`, which should be provided by `libboost_python`. Is any of the mentioned libs linked against this?

lucarinelli, 2020-01-22:
> Do the unit tests (e.g. moveit_commander/test/python_moveit_commander.test) actually work?

I'm not sure I have done the test correctly, I used `~/ws_moveit/src/moveit/moveit_commander$ catkin run_tests --no-deps --this -iv`.
It says fail and I see the same ImportError multiple times.

Here the entire output: https://gist.github.com/lucarinelli/35e3ae4f6126421a8174e995be2af494

rhaschke, 2020-01-22:
This means that the unit tests fail as well. This might be related to Ubuntu 19.10. Did you build your whole ROS from scratch (because there are no binary repos available, are they?)
Did you check the linked boost python library?

lucarinelli, 2020-01-22:
> Did you build your whole ROS from scratch (because there are no binary repos available, are they?)

No binary repository available, I built also ROS Melodic from scratch following the instructions of the official tutorial. Other components of MoveIt are working, I can plan trajectories with Rviz for example.

> Did you check the linked boost python library?

I haven't checked which is the boost python linked library, I'm still looking for a way to do that. Could you help me suggesting how to do that properly?

rhaschke, 2020-01-22:
```
ldd /home/lrinelli/ws_moveit/install/lib/libmoveit_py_bindings_tools.so.1.0.1
ldd /home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_roscpp_initializer.so
```
will list all libraries that are linked by these libs. Please report the result.

lucarinelli, 2020-01-22:
```
lrinelli@SAN22389:~$ ldd /home/lrinelli/ws_moveit/install/lib/libmoveit_py_bindings_tools.so.1.0.1
	linux-vdso.so.1 (0x00007fff38773000)
	libpython2.7.so.1.0 => /lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f5997e94000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f5997e71000)
	libroscpp.so => not found
	libboost_system.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_system.so.1.67.0 (0x00007f5997e6a000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f5997c7c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f5997c62000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5997a6f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f5997a53000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5997a4d000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f5997a48000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f59978f9000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f599826f000)
lrinelli@SAN22389:~$ ldd /home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_roscpp_initializer.so
	linux-vdso.so.1 (0x00007ffd07f8c000)
	libmoveit_py_bindings_tools.so.1.0.1 => not found
	libpython2.7.so.1.0 => /lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f2ff38db000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2ff36ed000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2ff36d3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2ff34e2000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f2ff34c6000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2ff34a1000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2ff349b000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f2ff3496000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2ff3347000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2ff3c9e000)
```

rhaschke, 2020-01-22:
Indeed libboost_python is not linked. It should look like this:
```
$ ldd /opt/ros/melodic/lib/libmoveit_py_bindings_tools.so
	linux-vdso.so.1 (0x00007ffc5fdfa000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f2466ef0000)
	libboost_python-py27.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1 (0x00007f2466cb0000)
        ...
```
Could you please try to change:
https://github.com/ros-planning/moveit/blob/1389df26358746d123306915fe1751616d09dc34/moveit_ros/planning_interface/CMakeLists.txt#L31-L36
to use `LESS_EQUAL` instead of `LESS`:
`if(Boost_VERSION LESS_EQUAL 106700)`

If this works, please file a PR.

lucarinelli, 2020-01-23:
boost::python seems to be there

```
~$ dpkg -l | grep boost | grep python
ii  libboost-mpi-python-dev                    1.67.0.2                                  amd64        C++ interface to the Message Passing Interface (MPI), Python Bindings (default version)
ii  libboost-mpi-python1.67-dev                1.67.0-13ubuntu1                          amd64        C++ interface to the Message Passing Interface (MPI), Python Bindings
ii  libboost-mpi-python1.67.0                  1.67.0-13ubuntu1                          amd64        C++ interface to the Message Passing Interface (MPI), Python Bindings
ii  libboost-python-dev                        1.67.0.2                                  amd64        Boost.Python Library development files (default version)
ii  libboost-python1.67-dev                    1.67.0-13ubuntu1                          amd64        Boost.Python Library development files
ii  libboost-python1.67.0                      1.67.0-13ubuntu1                          amd64        Boost.Python Library
```

```
/usr/lib/x86_64-linux-gnu$ ls libboost_python*
libboost_python27.a          libboost_python37.so         libboost_python3-py37.so
libboost_python27.so         libboost_python37.so.1.67.0  libboost_python3.so
libboost_python27.so.1.67.0  libboost_python3.a           libboost_python.a
libboost_python37.a          libboost_python3-py37.a      libboost_python.so
```
Running `catkin clean` and then `~/ws_moveit$ catkin build --force-cmake -DCMAKE_CXX_FLAGS=-Wall -DCMAKE_C_FLAGS=-Wall moveit_commander` I have no warning related to boost, python or linked libraries.

I think I found a typo in those five lines #1857.

Now libboost_python is being found and linked:
```
lrinelli@SAN22389:/usr/lib/x86_64-linux-gnu$ ldd /home/lrinelli/ws_moveit/install/lib/libmoveit_py_bindings_tools.so.1.0.1 | grep python
	libpython2.7.so.1.0 => /lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f6c8997e000)
	libboost_python27.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_python27.so.1.67.0 (0x00007f6c89931000)
lrinelli@SAN22389:/usr/lib/x86_64-linux-gnu$ ldd /home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_roscpp_initializer.so | grep python
	libpython2.7.so.1.0 => /lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f6821fbf000)
	libboost_python27.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_python27.so.1.67.0 (0x00007f6821f79000)
```

Now I get

```
Traceback (most recent call last):
  File "./test_print_state.py", line 6, in <module>
    import moveit_commander
  File "/home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_commander/__init__.py", line 4, in <module>
    from .move_group import *
  File "/home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 41, in <module>
    from moveit_ros_planning_interface import _moveit_move_group_interface
ImportError: /home/lrinelli/ws_moveit/install/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_move_group_interface.so: undefined symbol: _ZN7eigenpy13enableEigenPyEv

```

rhaschke, 2020-01-23:
Thanks for the fix. The next issue is related to `eigenpy`:
https://github.com/ros-planning/moveit/blob/e582953c8c7a92f19674479f9986fd45d44d4f92/moveit_ros/planning_interface/CMakeLists.txt#L47

https://github.com/ros-planning/moveit/blob/e582953c8c7a92f19674479f9986fd45d44d4f92/moveit_ros/planning_interface/move_group_interface/CMakeLists.txt#L9
 Again, please check whether you have eigenpy installed and whether linking works as expected:
```
$ ldd /opt/ros/melodic/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_move_group_interface.so | grep py
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007fce2c236000)
	libeigenpy.so => /opt/ros/melodic/lib/libeigenpy.so (0x00007fce2b61b000)
	libboost_python-py27.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1 (0x00007fce2b3db000)
	libmoveit_py_bindings_tools.so.1.0.2 => /opt/ros/melodic/lib/libmoveit_py_bindings_tools.so.1.0.2 (0x00007fce2a5c3000)
```

lucarinelli, 2020-01-23:
I'm not sure I'm installing eigenpy in the proprer way, i'm just
- cloning https://github.com/stack-of-tasks/eigenpy master in `ws_moveit/src`
- initializing submodules
- `catkin build` (after having done `catkin config --install`)

By the way `ldd` does not shows `libeigenpy.so` and `find_package(eigenpy REQUIRED)` does not fail or give any error, but the variables `eigenpy_LIBRARIES` and `eigenpy_INCLUDE_DIRS` are not set.

This is the list of the variables set with eigenpy in the name:
```
-- CONFIG_FILES=/home/lrinelli/ws_moveit/install/lib/cmake/eigenpy/eigenpyTargets-noconfig.cmake
-- EIGENPY_DOXYGENDOCDIR=/home/lrinelli/ws_moveit/install/share/doc/eigenpy/doxygen-html
-- _DIR=/home/lrinelli/ws_moveit/install/lib/cmake/eigenpy
-- eigenpy_CONFIG=/home/lrinelli/ws_moveit/install/lib/cmake/eigenpy/eigenpyConfig.cmake
-- eigenpy_CONSIDERED_CONFIGS=/home/lrinelli/ws_moveit/install/lib/cmake/eigenpy/eigenpyConfig.cmake
-- eigenpy_CONSIDERED_VERSIONS=1.6.13
-- eigenpy_DEPENDENCIES=
-- eigenpy_DIR=/home/lrinelli/ws_moveit/install/lib/cmake/eigenpy
-- eigenpy_DOXYGENDOCDIR=/home/lrinelli/ws_moveit/install/share/doc/eigenpy/doxygen-html
-- eigenpy_FOUND=1
-- eigenpy_PKG_CONFIG_DEPENDENCIES=eigen3 >= 3.0.5
-- eigenpy_VERSION=1.6.13
-- eigenpy_VERSION_COUNT=3
-- eigenpy_VERSION_MAJOR=1
-- eigenpy_VERSION_MINOR=6
-- eigenpy_VERSION_PATCH=13
-- eigenpy_VERSION_TWEAK=0
```

rhaschke, 2020-01-23:
I see. eigenpy uses a customized release process to ensure compatibility to normal cmake files (which the original package doesn't do). Hence try to use:
https://github.com/ipab-slmc/eigenpy_catkin-release/tree/release/melodic/eigenpy/1.6.9-1

lucarinelli, 2020-01-23:
Ok, that solves the problem!
Thank you!

closed 2020-01-23 by rhaschke