moveit/moveit #1864: moveit_chomp_optimizer_adapter fails to build on Kinetic

tfoote, 2020-01-23:
From: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__moveit_chomp_optimizer_adapter__ubuntu_xenial_amd64__binary/209/console

It looks like a missing dependency.

```
00:04:25.831 make[2]: Entering directory '/tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu'
00:04:25.831 /usr/bin/cmake -H/tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17 -B/tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu --check-build-system CMakeFiles/Makefile.cmake 0
00:04:25.845 /usr/bin/cmake -E cmake_progress_start /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu/CMakeFiles /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu/CMakeFiles/progress.marks
00:04:25.855 make -f CMakeFiles/Makefile2 all
00:04:25.859 make[3]: Entering directory '/tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu'
00:04:25.859 make -f CMakeFiles/moveit_chomp_optimizer_adapter.dir/build.make CMakeFiles/moveit_chomp_optimizer_adapter.dir/depend
00:04:25.861 make[4]: Entering directory '/tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu'
00:04:25.861 cd /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17 /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17 /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu/CMakeFiles/moveit_chomp_optimizer_adapter.dir/DependInfo.cmake --color=
00:04:25.918 Scanning dependencies of target moveit_chomp_optimizer_adapter
00:04:25.920 make[4]: Leaving directory '/tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu'
00:04:25.920 make -f CMakeFiles/moveit_chomp_optimizer_adapter.dir/build.make CMakeFiles/moveit_chomp_optimizer_adapter.dir/build
00:04:25.932 make[4]: Entering directory '/tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/obj-x86_64-linux-gnu'
00:04:25.941 [ 50%] Building CXX object CMakeFiles/moveit_chomp_optimizer_adapter.dir/src/chomp_optimizer_adapter.cpp.o
00:04:25.942 /usr/lib/ccache/x86_64-linux-gnu-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"moveit_chomp_optimizer_adapter\" -Dmoveit_chomp_optimizer_adapter_EXPORTS -I/opt/ros/kinetic/include -I/opt/ros/kinetic/share/orocos_kdl/../../include -I/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/usr/include/eigen3  -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2  -fPIC   -std=c++14 -o CMakeFiles/moveit_chomp_optimizer_adapter.dir/src/chomp_optimizer_adapter.cpp.o -c /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/src/chomp_optimizer_adapter.cpp
00:04:29.267 /tmp/binarydeb/ros-kinetic-moveit-chomp-optimizer-adapter-0.9.17/src/chomp_optimizer_adapter.cpp:51:35: fatal error: tf/transform_listener.h: No such file or directory
00:04:29.267 compilation terminated.
```

rhaschke, 2020-01-24:
Thanks for reporting. I already fixed it and running a prerelease test now:
https://travis-ci.com/ubi-agni/moveit/jobs/279635070

rhaschke, 2020-01-24:
Should be fixed via https://github.com/ros/rosdistro/pull/23593.

closed 2020-01-24 by rhaschke