moveit/moveit #1878: prefer `moveit::core` over `robot_state` namespace

v4hn, 2020-02-05:
Confusingly, [robot_model/robot_model.h](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_model/include/moveit/robot_model/robot_model.h#L621) creates alias namespaces for `moveit::core`. Ioan added this long ago to stay backward compatible with old usage.

However, as a result the namespaces where not aligned throughout the codebase (grep for `robot_state::RobotState`, for example) and the alias is still there.

I believe we should deprecate the old names and align our codebases not to use the old names.

@davetcoleman

davetcoleman, 2020-06-02:
This duplicate namespacing has always bothered me. @v4hn why did you close this? How about we make the change now in MoveIt 2? @henningkayser

rhaschke, 2020-06-02:
Actually, most of these namespace instances were removed in #1924. But there are still a few left...

v4hn, 2020-06-02:
I was under the impression the linked pr fixed it, but @rhaschke is right, apparently there are still some places left..

v4hn, 2020-06-02:
pr will be ready in 5 :-)

PR #2135 (2020-06-03): remove robot_model:: and robot_state::
Fixes #1878 .

Please merge this as two commits, as we should only backport the internal usage, but not the API change.
closed 2020-06-03 by rhaschke