moveit/moveit #1888: default velocity / acceleration factors

v4hn, 2020-02-10:
I thought I already created an issue for this at some point, but apparently not. So here it goes...

A usual problem with robot configurations is that maintainers decide to specify severely reduced velocity/acceleration limits.
Of course, this ensures that the robot will never run into anything at high speeds when given a very simple motion command.
This is clearly appreciated for new users and makes sense as default behavior!

But it also means that there is no easy way to increase the speed later on without modifying the per-joint configuration (which should only be done with knowledge of the specs, of course).
In the past, I was asked quite a number of times, why MoveIt's trajectories are so slow, when "industry" demos with the same robot arms can move so much faster..
So this is actually bad advertisement for some people who might try it.

As we already have the concept of velocity/acceleration scaling factors, I would propose to add two simple parameters `default_velocity_scaling_factor` and `default_acceleration_scaling_factor` (to our joint_limits.yaml?) and support them throughout the pipeline.
This should allow the average user to speed up their trajectories simply by setting higher factors.

Any thoughts on this @rhaschke @gavanderhoorn ?

gavanderhoorn, 2020-02-10:
> As we already have the concept of velocity/acceleration scaling factors, I would propose to add two simple parameters `default_velocity_scaling_factor` and `default_acceleration_scaling_factor` (to our joint_limits.yaml?) and support them throughout the pipeline.

I'm not sure I completely understand what the idea is with these new parameters.

Could you clarify?

jschleicher, 2020-02-10:
Do you think of using demo.launch to reduce robot speed when run from RViz? We solved that by setting default values in the rviz config of [prbt_moveit_config](https://github.com/PilzDE/pilz_robots/tree/melodic-devel/prbt_moveit_config) to 20%. This value reduces the default for first-time users. move_group-requests (programmatically) can still be executed full speed and the user can increase the speed in the plugin as well. But the default(s) shouldn't be 1.0 of course.

[image]

Maybe it's also possible to reduce the default velocity/acceleration scaling factors inside MotionPlanRequest message so that one has to override them to run full speed?

I fully support your opinion, that the joint_limits.yaml / URDF limits shouldn't be messed with to reduce to safe speed at all. These values should always match the hardware limits.

rhaschke, 2020-02-10:
Thanks @jschleicher for proposing concrete options to implement this behavior. I agree with both of you, that this is a sensible and desirable approach. I started a corresponding PR in #1889.

felixvd, 2020-03-12:
Fixed by #1890

closed 2020-03-12 by v4hn