moveit/moveit #1896: Disable codecov as a prerequisite for github merging

rhaschke, 2020-02-13:
Dear @tylerjw, @davetcoleman,
while having the codecov report shown in github is great, can we relax the requirement to not decrease coverage with PRs? This creates lots of spurious "not-mergable" flags, I one always needs to inspect the details, why a particular PR is not mergable.
As we currently have extremely low coverage, I suggest to invert the messaging and provide extra positive feedback if the coverage is significantly increased, because a new test case was added.
Is that possible?

rhaschke, 2020-02-21:
Ping @tylerjw. We should discuss our strategy in the next maintainer meeting.
Could you, @tylerjw, figure out the technical possibilities until then?

felixvd, 2020-02-25:
Cross-referencing the [discourse post](https://discourse.ros.org/t/moveit-maintainer-meeting-feb-27th/12812/5?u=felixvd) for the meeting.

jschleicher, 2020-02-28:
See PR #1917

v4hn, 2020-03-20:
The remaining issue, as discussed at the last maintainer meeting is the red 'x' in every pull-request that fails codecov.

This is not wanted right now and makes the maintainers' job harder looking through all open requests.

jschleicher, 2020-04-22:
@v4hn I think, this was fixed in #1917, see current PR #2027 for a working example: coverage decreases and it is still green. Suggesting to close here.

closed 2020-04-23 by v4hn