moveit/moveit #1902: Eigen RobotState memory issue when linked with OpenVINO

tylerjw, 2020-02-14:
### Description

I created a minimal example that reproduces this bug here: https://github.com/tylerjw/moveit_memory_bug

OpenVINO is a library provided from intel for ML inference among other things.  If it is linked in (never called) then this short bit of code will generate a memory alignment Eigen error in the constructor for MoveitCpp.  Here is the code in the test:

```
#include <memory>
#include <moveit/moveit_cpp/moveit_cpp.h>
#include <gtest/gtest.h>
#include <ros/ros.h>

namespace {
// Comment out this function that is never called to see the bug go away
void test_function(const planning_scene_monitor::PlanningSceneMonitorPtr
                       &planning_scene_monitor) {
  const moveit::core::RobotStatePtr &kinematic_state =
      planning_scene_monitor->getStateMonitor()->getCurrentState();
  const Eigen::Isometry3d &command_to_camera_frame_transform =
      kinematic_state->getGlobalLinkTransform("base_link").inverse() *
      kinematic_state->getGlobalLinkTransform("camera");
}
}

TEST(ExampleServerTest, ExampleUnitTests) {
  ros::NodeHandle nh("~");
  moveit::planning_interface::MoveItCpp::Options moveit_cpp_options(nh);

  // Initialize MoveIt planning stack using MoveItCpp
  std::shared_ptr<moveit::planning_interface::MoveItCpp> moveit_cpp =
      std::make_shared<moveit::planning_interface::MoveItCpp>(
          moveit_cpp_options, nh);
}

int main(int argc, char **argv) {
  ros::init(argc, argv, "unit_test");
  ::testing::InitGoogleTest(&argc, argv);
  return RUN_ALL_TESTS();
}
```

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source build, master branch

### Backtrace/Console output
```
unit_test: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 16; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

Thread 1 "unit_test" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51  ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff5683801 in __GI_abort () at abort.c:79
#2  0x00007ffff567339a in __assert_fail_base (fmt=0x7ffff57fa7d8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x5555555c8150 "(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.htm"..., 
    file=file@entry=0x5555555c8118 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=line@entry=128, 
    function=function@entry=0x5555555c8c00 <Eigen::internal::plain_array<double, 16, 0, 32>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 16; int MatrixOrArrayOptions = 0]")
    at assert.c:92
#3  0x00007ffff5673412 in __GI___assert_fail (
    assertion=0x5555555c8150 "(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.htm"..., 
    file=0x5555555c8118 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=128, 
    function=0x5555555c8c00 <Eigen::internal::plain_array<double, 16, 0, 32>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 16; int MatrixOrArrayOptions = 0]") at assert.c:101
#4  0x00005555555b42cd in Eigen::internal::plain_array<double, 16, 0, 32>::plain_array (this=0x55555585ae70)
    at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128
#5  0x00005555555b3952 in Eigen::DenseStorage<double, 16, 4, 4, 0>::DenseStorage (this=0x55555585ae70)
    at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:187
#6  0x00005555555b28da in Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 4, 0, 4, 4> >::PlainObjectBase (this=0x55555585ae70)
[... 65 lines cut ...]
#36 0x00005555555adc31 in ExampleServerTest_ExampleUnitTests_Test::TestBody (this=0x55555583f220)
    at /home/tyler-pick/workspace/ws_bug/src/moveit_memory_bug/test/unit_test.cpp:37
#37 0x00007ffff799ac03 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=0x55555583f220, 
    method=&virtual testing::Test::TestBody(), location=0x7ffff79aad93 "the test body") at /usr/src/googletest/googletest/src/gtest.cc:2402
#38 0x00007ffff7994b8b in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=0x55555583f220, 
    method=&virtual testing::Test::TestBody(), location=0x7ffff79aad93 "the test body") at /usr/src/googletest/googletest/src/gtest.cc:2438
#39 0x00007ffff79780f4 in testing::Test::Run (this=0x55555583f220) at /usr/src/googletest/googletest/src/gtest.cc:2474
#40 0x00007ffff7978a2e in testing::TestInfo::Run (this=0x55555583a530) at /usr/src/googletest/googletest/src/gtest.cc:2656
#41 0x00007ffff79790d1 in testing::TestCase::Run (this=0x555555816c00) at /usr/src/googletest/googletest/src/gtest.cc:2776
#42 0x00007ffff798007a in testing::internal::UnitTestImpl::RunAllTests (this=0x55555583a780)
    at /usr/src/googletest/googletest/src/gtest.cc:4651
#43 0x00007ffff799bdc9 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (
    object=0x55555583a780, 
    method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x7ffff797fda8 <testing::internal::UnitTestImpl::RunAllTests()>, location=0x7ffff79ab5d0 "auxiliary test code (environments or event listeners)")
    at /usr/src/googletest/googletest/src/gtest.cc:2402
#44 0x00007ffff79959c7 in testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (
    object=0x55555583a780, 
    method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x7ffff797fda8 <testing::internal::UnitTestImpl::RunAllTests()>, location=0x7ffff79ab5d0 "auxiliary test code (environments or event listeners)")
    at /usr/src/googletest/googletest/src/gtest.cc:2438
#45 0x00007ffff797ec0b in testing::UnitTest::Run (this=0x7ffff7bcd7c0 <testing::UnitTest::GetInstance()::instance>)
    at /usr/src/googletest/googletest/src/gtest.cc:4259
#46 0x00005555555b01fe in RUN_ALL_TESTS () at /usr/src/googletest/googletest/include/gtest/gtest.h:2233
#47 0x00005555555add7b in main (argc=2, argv=0x7fffffffcbe8)
    at /home/tyler-pick/workspace/ws_bug/src/moveit_memory_bug/test/unit_test.cpp:43f
```

rhaschke, 2020-02-15:
This is an old bug: #1385, which has a known but labor-intensive solution.
Unfortunately, it was never considered urgent (as some other recently introduced regressions, e.g. #1835 - interesting coincidence of digits in issue numbers :wink:)

tylerjw, 2020-02-15:
@rhaschke In the meantime I think I'm going to have to isolate the OpenVINO code onto another node for the project where I found this.  At the same time I am a good person though to start going through MoveIt and modernizing our Eigen usage.  

Is there a specific place I should start?  Would the RobotState be a good place to start?

Also, is there a reason we use normal Eigen instead of MLK in MoveIt?  My understanding is that because we are using the BSD license we can take advantage of MKL (https://eigen.tuxfamily.org/dox/TopicUsingIntelMKL.html).

rhaschke, 2020-02-15:
I think, RobotState I already adapted appropriately in #1382. What is missing, are all kind of Eigen object instances within structures. Best, start with your particular issue and try to find a fix for it - looks like a good start is `LinkModel`.

mamoll, 2020-02-15:
@tylerjw OMPL has it's own Eigen issues: https://github.com/ompl/ompl/issues/601. I have [made some fixes](https://github.com/ompl/ompl/commit/10c847d34eaa3deb244ea5a4ecdee406e1778326), but I am not sure all issues have been resolved.