moveit/moveit #1903: why 'rosdep install' install some moveit binary package when i installed moveit from source?

jsbyysheng, 2020-02-15:
### Description
I followed the direction of https://moveit.ros.org/install/source/ and ran commands step by step. When running `rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}`, I noticed it started to install `ros-melodic-panda-moveit-config` and also some additional packages like `ros-melodic-moveit-core`, `ros-melodic-moveit-ros-planning`, etc. I was confused why these packages were installed while I had added `--ignore-src` ?

Overview of your issue here.

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source, master-branch 

### Steps to reproduce
Following the direction of https://moveit.ros.org/install/source/ step by step.
When getting `rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}`.

### Expected behaviour
Some additional packages like `ros-melodic-moveit-core`, `ros-melodic-moveit-ros-planning`, etc have been installed.

### Actual behaviour
Additional packages like `ros-melodic-moveit-core`, `ros-melodic-moveit-ros-planning`, etc should not been installed.

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2020-02-15:
Probably you changed folders before executing `rosdep install`.
This command needs to be executed in the root folder of your catkin workspace.
As you can see from our regular CI builds, performing exactly the same command, no MoveIt packages are installed in this case:
https://travis-ci.org/ros-planning/moveit/jobs/650398586#L512-L597.

jsbyysheng, 2020-02-15:
> Probably you changed folders before executing `rosdep install`.
> This command needs to be executed in the root folder of your catkin workspace.
> As you can see from our regular CI builds, performing [...]

I am very glad for your reply but I still think it's a bug in `moveit.rosinstall` file. Currently, this file lacks   `ros-planning/panda_moveit_config`,  which is required by `moveit/moveit_ros/benchmarks/examples` and some other places. So, when I executed `rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}`, rosdep starts to install `ros-melodic-panda_moveit_config`, which depends on other packages. After I add `ros-planning/panda_moveit_config` to the `moveit.rosinstall` file, all works well.

I also have read the CI builds file you mentioned before. I wonder if `ros-planning/panda_moveit_config` has contained in the docker images? I am very sorry I cannot pull the image to verify it because of the bad internet environment.

rhaschke, 2020-02-15:
Thanks for clarifying your actual issue. Actually, MoveIt shouldn't depend on `panda_moveit_config` in first place. This is only required for the tutorials.

PR #1904 (2020-03-09): Fix usages of panda_moveit_config
As pointed out in https://github.com/ros-planning/moveit/issues/1903#issuecomment-586569789, following our [install-from-source procedure](https://moveit.ros.org/install/source/), will install `panda_move_config` and, as their dependencies, several other MoveIt packages from debians.
This is usually not desired.
Interestingly, none of the MoveIt packages actually package-depend on panda_moveit_config.
However, `moveit_tutorials` does. As this, but not `panda_moveit_config` is listed in `moveit.rosinstall`, all these binary packages get installed.
Searching for [... cut]
closed 2020-03-09 by rhaschke