moveit/moveit #1908: Name of method setJointPositions in robot_state is misleading

SammyRamone, 2020-02-20:
### Description

The method "setJointPositions" in robot_state is misleading. Due to the "s" at the end one assumes that all joints are set if provided with names and positions. In fact only one joint is set.
The method works, it's only misleading due to the name. Maybe think about removing the last "s", or make it set multiple joints.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build 1.0.2-1bionic.20200130.024855

### Expected behaviour
Setting position of all joints.

### Actual behaviour
Setting position of just one joint.

v4hn, 2020-02-20:
Hi Marc, thanks for the feedback.

For reference, [here's the relevant prototypes](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L493).

I do understand the confusion. Indeed, there is no overload of the `setJointPositions` method that touches more than one joint... This can be puzzling for new users for sure.

However, the naming is consistent throughout the class, because the methods can set **multiple positions** of the **same** joint (for joints with more than one DoF). There is also the family of `setJointGroupPositions` which follow the same naming schema.

The method you were looking for would be [any overload of `setVariablePositions`](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L175) instead.

It probably makes sense to add a sentence on this to the doxygen documentation?
Could you/ whoever noticed this create a pull-request for this?

v4hn, 2020-03-13:
Resolved by #1921. Again, thanks for reporting @SammyRamone

closed 2020-03-13 by v4hn