moveit/moveit #1913: Connect Moveit with KUKA

CesMak, 2020-02-25:
Hey there,

I successfully connected my 30HA-KUKA robot using [kuka_eki_hw_interface](https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_eki_hw_interface) to my Ubuntu 18.04 melodic ROS PC. 

I am now able to control the robot using
` rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller`

**Goal**
I would like to control the EE of the real Robot using MoveIt Plugin in Rviz.

**Progress**
I read in [Low-Level-Controllers-MOVEIT](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html) that a FollowJointTrajectory action service is required. 

If I do rostopic echo (when connected to the real Robot with kuka_eki_hw_interface) I get:

```
/joint_states
/position_trajectory_controller/command
/position_trajectory_controller/follow_joint_trajectory/cancel
/position_trajectory_controller/follow_joint_trajectory/feedback
/position_trajectory_controller/follow_joint_trajectory/goal
/position_trajectory_controller/follow_joint_trajectory/result
/position_trajectory_controller/follow_joint_trajectory/status
/position_trajectory_controller/state
/rosout
/rosout_agg
/tf
/tf_static
```

-> So this means I have an action server as required already right?

Next I just have to make sure, that I remap the above topics to these ones or (as explained in the tutorial) :

```
        /panda_arm_controller/follow_joint_trajectory/goal
        /panda_arm_controller/follow_joint_trajectory/feedback
        /panda_arm_controller/follow_joint_trajectory/result
        /hand_controller/gripper_action/goal
        /hand_controller/gripper_action/feedback
        /hand_controller/gripper_action/result
```

felixvd, 2020-02-26:
This question would be better suited for and probably get more views on [ROS answers](https://answers.ros.org/questions/).

To give a stub of a response: I am confused why you start up the panda's moveit config although you are using a KUKA robot. You should probably start your troubleshooting there. Do take it over to ROS answers though. You will get better responses.

gavanderhoorn, 2020-02-26:
Unfortunately the response to MoveIt related questions on ROS Answers has been really low. It makes little sense to direct people to ROS Answers when none of the MoveIt maintainers actually monitor that forum.

I can understand why users start posting these sort of questions here because of that.

I realise it's not just the maintainers that should answer questions there, but it would seem the reality is that either MoveIt users with sufficient experience to answer these types of questions do not frequent ROS Answers, or there are no MoveIt users on ROS Answers, period.

felixvd, 2020-02-27:
I also don't judge anyone for it, and it's not certain every answer will get a better response over there, but at least it is better to put it in a central place to search.

CesMak, 2020-03-03:
moved and **updated** Qustion to:

https://answers.ros.org/question/345728/connect-real-kuka-to-moveit/

tylerjw, 2020-04-09:
@CesMak was your issue solved?  Should this be closed?

CesMak, 2020-04-10:
Yes can be closed

closed 2020-04-10 by rhaschke