moveit/moveit #1914: moveit_jog_arm fails to build

m-talha, 2020-02-25:
### Description

I am trying to build `moveit_jog_arm` from master following the instructions [here](https://moveit.ros.org/install/source/). However it fails due to `fatal error: moveit_msgs/ChangeDriftDimensions.h: No such file or directory`. Should note that `moveit/msgs` builds fine. Just for some reason, the header files are in the `devel` folder which I presume CMake does not about.

I did manage to get it to compile correctly with the install flag but then sourcing the resulting setup file didn't do anything.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? Source- Master

### Steps to reproduce
Follow the instructions above.

JafarAbdi, 2020-02-25:
@m-talha Did you install MoveIt from source by following these steps https://moveit.ros.org/install/source/ .?

The `ChangeDriftDimensions` service isn't available in the lastest melodic release, you must install https://github.com/ros-planning/moveit_msgs from source, too

m-talha, 2020-02-25:
Yes, as I said above, I got all my code from source. I can see the service source file as well as the generated header file. However, the header file is in `devel/include` which I think is not available to CMake when it is building `moveit_jog_arm`.

m-talha, 2020-02-26:
Ok so it looks like the issue has resolved itself. 

I deleted everything and started from scratch again and it worked without a problem this time. Strange

closed 2020-02-26 by m-talha