moveit/moveit #1916: Fail: RECALLED: This goal was canceled because another goal was recieved by the simple action server

lusu8892, 2020-02-25:
### Description

I am running my code, Moveit and Rviz on the remote PC (Master PC), this PC is running roscore also. The hardware is paired with another PC (aka Hardware PC). My code will use move_group to call pick function to plan a object picking planning. However I got the warning, **Fail: RECALLED: This goal was canceled because another goal was recieved by the simple action server.**

But if I run my code, Moveit and Rviz on Hardware PC, but keeping roscore on the Master PC, everything works fine.

### Your environment
* ROS Distro: [Kinetic|]
* OS Version: All PCs running Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

v4hn, 2020-02-26:
Your code calls the action twice for some reason, while one call is still processed.
`rostopic echo /pickup/goal` to monitor the requests you send.

Almost surely, this is a bug in your setup.

closed 2020-02-26 by v4hn

lusu8892, 2020-02-26:
Hi v4hn

I checked the topic, I can not target where did I send the same message to the action server.

I tried running debug mode. the pick_action_client_ only sendGoal once, I do not know where I sent it twice.

v4hn, 2020-02-27:
Sorry but this seems to be close to impossible to debug or resolve without access to your specific setup.