moveit/moveit #1918: Can'

tylerjw, 2020-02-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Dale-Koenig, 2020-02-28:
catkin compiles to install rather than normal build, ros tools seem to try to take you to the install directory instead.  so `rostest` will look in the install directory and find no tests.

rhaschke, 2020-02-28:
That's correct. If you want to be able to run tests, you need to source the devel space, not the install space. Closing the issue.

closed 2020-02-28 by rhaschke