moveit/moveit #1920: approximate solutions from OMPL should be handled as failure

simonschmeisser, 2020-02-29:
### Description

I spent quite some time the last few days trying to figure out why some of our grasp plans contained jumps (the customer is very cautious and quick with emergency stop and as such no major (?) damage happened). Turns out that sometimes segments we were using MoveIt/ompl for contained almost no motion (~1e-4 joint difference between start and goal and never reached the goal). It then jumped to the next, cartesian, part of the grasp. We now have continuity checks ...

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source: melodic-devel, about 1.5 month old

### Steps to reproduce

hard, usually planning either fails or succeeds but sometimes approximate solutions are found 

### Expected behaviour

Approximate solutions should be considered a failure unless specifically requested, after all the goal is not reached

### Actual behaviour

a warning is printed ...

https://github.com/ros-planning/moveit/blob/fe39430d971fdfadcec0c7316fa67d41bf48b281/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L659

definition of "approximate" from OMPL:
https://ompl.kavrakilab.org/classompl_1_1base_1_1ProblemDefinition.html#a9b78738e28269c51f0f28103675fd831

@mamoll do you agree that approximate solutions should be rejected or would you rather expand the MoveIt interface (https://github.com/ros-planning/moveit_msgs/blob/master/msg/MoveItErrorCodes.msg) to expose this information?

v4hn, 2020-03-01:
The almost empty "approximate solutions" you describe actually sound like an issue inside OMPL, but that's for them to judge.

> if the top found solution is approximate (does not actually reach the desired goal, but hopefully is closer to it) 

Given this description, I'm horrified to see we accept such solutions with only a warning!
While it might make sense to keep the warning around, planning should be considered failed if it does not satisfy the goal constraints. Can you provide a pull-request?

simonschmeisser, 2020-03-02:
I added some further debug output and there are cases were the startState of the returned trajectory has a distance of 7 or greater from the first point of the trajectory!

Adding  some low-level validity filters is complicated further by the last waypoint always having a distance > 0 to the goal despite the goal being a joint value target (ie should be considered fixed)

simonschmeisser, 2020-03-02:
I'll open an OMPL issue later on as well as the returned "approximate trajectory" goes from somewhere close to the goal to the goal instead of from the start towards somewhere closer to the goal.

This seems to happen when there are pretty extreme movements involved, see the following example:
[image]
Here basically the robot needs to do a full flip.

We're using RRTConnect.

v4hn, 2020-03-03:
> Adding  some low-level validity filters is complicated further by the last waypoint always having a distance > 0 to the goal despite the goal being a joint value target (ie should be considered fixed)

Did you explicitly specify 0 tolerances for the targets?

simonschmeisser, 2020-03-04:
> Did you explicitly specify 0 tolerances for the targets?

No, will try if that works

simonschmeisser, 2020-03-04:
Thanks for the side hint, setting the goal_joint_tolerance to 0 allows to have a much cleaner workaround

mamoll, 2020-03-09:
OMPL's planners return a status indicating what type of solution has been found: https://ompl.kavrakilab.org/structompl_1_1base_1_1PlannerStatus.html. There is a convenience bool cast operator that converts EXACT and APPROXIMATE solutions to true and the rest to false. MoveIt can check that status explicitly to distinguish between exact and approximate solutions.

v4hn, 2020-03-13:
@mamoll Could you prepare a patch to make MoveIt reject everything that is not an `EXACT` solution?

Given @simonschmeisser's experience, `APPROXIMATE` might be ok for OMPL in general, but robotic execution clearly requires `EXACT`.

mamoll, 2020-03-13:
I just checked the MoveIt code and it seems like it is already checking whether an `EXACT` solution is found. See, e.g., https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L750. I don't understand how you get `APPROXIMATE` solutions if ompl_interface::ModelBasedPlanningContext::solve returns `true`.

v4hn, 2020-03-15:
@simonschmeisser Please comment on @mamoll's observations.
Did you use your own/an non-standard interface for planning?

simonschmeisser, 2020-03-21:
Sorry for the slow response, the workaround is doing its job ...

We use the normal planning-pipeline interface which should end up in the place @mamoll pointed too. As we set `num_planning_attempts` to 3, we end up in this branch https://github.com/ros-planning/moveit/blob/f18fa5fb4c9b4405dff36f1dbdb5b10a84b08e25/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L675 actually but it still compares to `EXACT_SOLUTION`

So I'm following two ideas currently:

- comparison problems like here: https://github.com/ros-planning/moveit/pull/670 (but does not sound likely)
- solutions being mislabeled in OMPL

without having checked the whole code:

a solution being labeled as APPROXIMATE in ParallelPlan depends on `solutions_[0]` being approximate. Hybridization happens afterwards however and might change the result?

So after studying this for a while I claim this to be a bug in either PathHybridization (a), ParallelPlan (b) or RRTconnect (c) depending on preference. So here is my hypothesis:

Three planners are started and solution 0 is exactlyValid (not approximate)

solution 1 is approximate and includes only the section leading to the goal but not actually the start (c)

both are feed into PathHybridization as Plan (discarding information about solution 1 being an approximate solution) (b)

PathHybridization prefers the approximate solution and only checks that it reaches the goal, but not whether it starts at the start  (a)

ParallelPlan returns the hybridized solution and because solution 0 is exact, claims the hybridized solution to be exact as well.  (b)

(MoveIts PlanningPipeline calls isPathValid but does not check the start either)

What do you think @mamoll ?

simonschmeisser, 2020-03-21:
Also if it were working this approach actually does not sound ~~very intelligent~~ optimal as only the solution path is used and not the trees of both solutions.

 Is there an (non-optimizing) algorithm in OMPL that tries connecting all branches of trees growing from both start and goal? Like pRRTConnect? (We're currently on melodic so multi-query planners are not an option yet)

v4hn, 2020-03-21:
I'll leave the main response to @mamoll , just one addition.

> MoveIts PlanningPipeline calls isPathValid but does not check the start either

No and I'm not sure we need such a check there (although it could be implemented in a PRA).
The check already exists in the actual execution, but that would of course only trigger if the trajectories are executed through MoveIt too.

mamoll, 2020-03-23:
Do you have approximate solutions that do start at the start state? Do you know which planner would produce such solutions?

simonschmeisser, 2020-03-23:
I'm using rrrconnect, not sure if it produces approximate solutions that start at the start as well.  But my checks never found any solution (after hybridization and in MoveIt) that doesn't reach the goal.

simonschmeisser, 2020-05-14:
Maybe the `approximate solution` was a red hering as we are testing a couple of thousand possible picks I might have mixed up output from separate solutions. However now I'm back to this setup and currently get the message even while planning a single move

```
[DEBUG] [1589473333.595789965]: Using 'JointModel' parameterization for solving problem
[DEBUG] [1589473333.595858352]: Reusing cached planning context
[DEBUG] [1589473333.596175120]: manipulator[RRTConnectkConfigDefault]: Setting planning volume (affects SE2 & SE3 joints only) to x = [-5,000000, 5,000000], y = [-5,000000, 5,000000], z = [-5,000000, 5,000000]
[DEBUG] [1589473333.596354287]: Constructed a ConstrainedGoalSampler instance at address 0x7f30dc58fcd0
[DEBUG] [1589473333.596389178]: Starting goal sampling thread
[DEBUG] [1589473333.596463254]: Waiting for space information to be set up before the sampling thread can begin computation...
[ INFO] [1589473333.596934314]: Planner configuration 'manipulator[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[DEBUG] [1589473333.596992866]: The value of parameter 'longest_valid_segment_fraction' is now: '0.002'
[DEBUG] [1589473333.597043359]: The value of parameter 'range' is now: '0.1'
[DEBUG] [1589473333.597094393]: manipulator[RRTConnectkConfigDefault]: New planning context is set.
[DEBUG] [1589473333.597127549]: manipulator[RRTConnectkConfigDefault]: Solving the planning problem 4 times...
[DEBUG] [1589473333.606590486]: Beginning sampling thread computation
[DEBUG] [1589473333.610599725]: The value of parameter 'range' is now: '0.1'
[DEBUG] [1589473333.610735828]: The value of parameter 'range' is now: '0.1'
[DEBUG] [1589473333.610810499]: The value of parameter 'range' is now: '0.1'
[DEBUG] [1589473333.610869583]: The value of parameter 'range' is now: '0.1'
[DEBUG] [1589473333.611327005]: ParallelPlan.solveMore: starting planner manipulator/manipulator[RRTConnectkConfigDefault]
[DEBUG] [1589473333.613747733]: ParallelPlan.solveMore: starting planner manipulator/manipulator[RRTConnectkConfigDefault]
[DEBUG] [1589473333.613831599]: ParallelPlan.solveMore: starting planner manipulator/manipulator[RRTConnectkConfigDefault]
[DEBUG] [1589473333.614581336]: ParallelPlan.solveMore: starting planner manipulator/manipulator[RRTConnectkConfigDefault]
[DEBUG] [1589473333.616343718]: Adding goal state
[DEBUG] [1589473333.616684769]: Adding goal state
[DEBUG] [1589473333.616998948]: Adding goal state
[DEBUG] [1589473333.617305227]: Adding goal state
[DEBUG] [1589473333.617623679]: Adding goal state
[DEBUG] [1589473333.617912953]: Adding goal state
[DEBUG] [1589473333.618204301]: Adding goal state
[DEBUG] [1589473333.618496297]: Adding goal state
[DEBUG] [1589473333.618787010]: Adding goal state
[DEBUG] [1589473333.619080236]: Adding goal state
[DEBUG] [1589473333.619372113]: Adding goal state
[DEBUG] [1589473333.619675159]: Adding goal state
[DEBUG] [1589473333.619965269]: Adding goal state
[DEBUG] [1589473333.620036153]: manipulator[RRTConnectkConfigDefault]: Allocating a new state sampler (attempts to use path constraints)
[DEBUG] [1589473333.620079767]: manipulator[RRTConnectkConfigDefault]: Allocating default state sampler for state space
[ INFO] [1589473333.620115518]: manipulator/manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[DEBUG] [1589473333.620249419]: manipulator[RRTConnectkConfigDefault]: Allocating a new state sampler (attempts to use path constraints)
[DEBUG] [1589473333.620289850]: Adding goal state
[DEBUG] [1589473333.620333781]: manipulator[RRTConnectkConfigDefault]: Allocating default state sampler for state space
[ INFO] [1589473333.620381942]: manipulator/manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[DEBUG] [1589473333.620596066]: Adding goal state
[DEBUG] [1589473333.620887059]: Adding goal state
[DEBUG] [1589473333.620961827]: manipulator[RRTConnectkConfigDefault]: Allocating a new state sampler (attempts to use path constraints)
[DEBUG] [1589473333.621003208]: manipulator[RRTConnectkConfigDefault]: Allocating default state sampler for state space
[ INFO] [1589473333.621043083]: manipulator/manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[DEBUG] [1589473333.621181035]: Adding goal state
[DEBUG] [1589473333.621471901]: Adding goal state
[DEBUG] [1589473333.621766278]: Adding goal state
[DEBUG] [1589473333.622056778]: Adding goal state
[DEBUG] [1589473333.622351388]: Adding goal state
[DEBUG] [1589473333.622667431]: Adding goal state
[DEBUG] [1589473333.622979750]: Adding goal state
[DEBUG] [1589473333.623270408]: manipulator[RRTConnectkConfigDefault]: Allocating a new state sampler (attempts to use path constraints)
[DEBUG] [1589473333.623325057]: manipulator[RRTConnectkConfigDefault]: Allocating default state sampler for state space
[DEBUG] [1589473333.624332067]: Adding goal state
[ INFO] [1589473333.624391610]: manipulator/manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[DEBUG] [1589473333.624650194]: Adding goal state
[DEBUG] [1589473333.624948345]: Adding goal state
[DEBUG] [1589473333.900770452]: Adding goal state
[..]1000x
[DEBUG] [1589473333.901044959]: Adding goal state
[DEBUG] [1589473333.901324286]: Adding goal state
[DEBUG] [1589473333.901615643]: Adding goal state
[... cut ...]
[ INFO] [1589473343.693271003]: SimpleSetup: Path simplification took 0,022942 seconds and changed from 12 to 2 states
[DEBUG] [1589473343.693316597]: manipulator[RRTConnectkConfigDefault]: Returning successful solution with 2 states
[ WARN] [1589473343.693960593]: Trajectory does not start at start state, distance is  12.8419 , failing it.

```

Two states as returned by Path simplification is definitely wrong as we check direct interpolation outside OMPL (and the general MoveIt planning pipeline beforehand)

mikebentley15, 2022-01-13:
I am not using MoveIt, but I'm using RRTConnect from OMPL 1.5.0.  I can confirm that using RRTConnect, that sometimes (about once every 200 or 300 times), the approximate solution it returns is from the goal tree instead of the start tree, resulting in a totally unusable plan.

Takeaways:

1. This is a bug in OMPL's RRTConnect
2. You can filter out these answers by looking for returned approximate plans that start from a goal state

mikebentley15, 2022-01-13:
It looks like it was fixed in OMPL commit f25bb385ccb8584dd9b58cfd61bffdcc94cde0eb (https://github.com/ompl/ompl/pull/739) last March, but they haven't made a release yet containing that fix.

simonschmeisser, 2022-01-13:
Fixed in OMPL but waiting for a new release containing the fix

closed 2022-01-13 by simonschmeisser