moveit/moveit #1922: moveit_setup_assistant file dialog should start from current directory

AndyZe, 2020-03-02:
When browsing to an existing package, the setup assistant used to start from the current working directory. It would be nice to restore that behavior because it's a good time saver.

[image]

rhaschke, 2020-03-02:
This was changed in #1618 by @JafarAbdi. Obviously, the "security fix" suggested in #1497, doesn't yield the correct path. Actually, `QDir::currentPath()` yields the current application path, i.e. the last path explicitly set. While we don't explicitly set a path, ROS internally does when searching for packages. But even fetching the path directly after start, just yields the ROS default path (~/.ros).
Hence, my suggestion is to revert #1618.

felixvd, 2020-03-03:
Just want to add that `roslaunch X_moveit_config setup_assistant.launch` fills in the path for that package and is a great timesaver as well.

jschleicher, 2020-03-06:
Fixed in #1932. Please Close.

closed 2020-03-06 by AndyZe