moveit/moveit #1926: Moveit_commander.roscpp_initialize throws C++ No registered converter (TypeError) during ptvsd debugging

rickstaa, 2020-03-04:
### Problem description

Just for future reference. When trying to debug a script, which contains the moveit_commander.roscpp_initialize method, using the [ptvsd debugger](https://github.com/Microsoft/ptvsd) I get the following error:

```
Exception has occurred: TypeError
No registered converter was able to produce a C++ rvalue of type std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > from this Python object of type unicode
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/roscpp_initializer.py", line 40, in roscpp_initialize
    _moveit_roscpp_initializer.roscpp_init("move_group_commander_wrappers", args2)
  File "/home/ricks/Development/panda_openai_sim_ws/src/panda_reach_training/sandbox/scripts/other/moveit_commander_test.py", line 15, in <module>
    moveit_commander.roscpp_initialize(sys.argv)
```

# Solution
The problem was caused by the way the PTVSD modifies the system argument vector `sys.argv`. When debugging a script, it puts the python executable as the first argument of the `sys.argv`. When debugging a ROS python script containing the `moveit_commander.roscpp_initialize` method you, therefore, have to remove the python executable from the `sys.argv`.

### Your environment
* **ROS Distro:** Melodic
* **OS Version: e.g.** Ubuntu 18.04
* **Binary build:** Binary (v1.0.2)

### Steps to reproduce
1. Open the [vscode](https://code.visualstudio.com/) IDE.
2. Create a `moveit_commander_test` script containing the following code:

```
#! /usr/bin/env python
import sys
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
from math import pi
from std_msgs.msg import String
from moveit_commander.conversions import pose_to_list

if __name__ == "__main__":

    rospy.init_node("move_group_python_interface_tutorial", anonymous=True)
    moveit_commander.roscpp_initialize(sys.argv)
```

3. Debug the script and watch the error appear.

rickstaa, 2020-03-04:
### Solution
The problem was caused by the way the PTVSD modifies the system argument vector `sys.argv`. When debugging a script, it puts the python executable as the first argument of the `sys.argv`. When debugging a ROS python script containing the `moveit_commander.roscpp_initialize` method you, therefore, have to remove the python executable from the `sys.argv`.

closed 2020-03-04 by rickstaa

drfuzzyness, 2020-03-16:
For future visitors, one ugly solution is to run code like this whenever you initialize `moveit_commander`.

```py
cleaned_args = [ a for a in sys.argv if not a.endswith("my_script_name_here.py") ]
moveit_commander.roscpp_initialize(cleaned_args)
```

rickstaa, 2020-03-17:
@drfuzzyness Thanks a lot for your solution. Thats more failproof than my quick and dirty solution in which i just replaced `sys.argv` with `[]`. Based on your solution, I now use:

```
import os
cleaned_args = [a for a in sys.argv if not os.path.basename(__file__) in os.path.basename(__file__)]
moveit_commander.roscpp_initialize(cleaned_args)
```

I am however unsure whether this works in all run cases.

danem, 2020-04-29:
What about using `rospy.myargv()` to get the cleaned arguments?

rhaschke, 2020-04-29:
@rickstaa, I cannot reproduce this issue (anymore?).
Is it still a problem with a recent version of vscode?

rickstaa, 2020-05-18:
@rhaschke Thanks for the update. I just performed some tests and this problem seems to be solved for vscode version 1.46.0-insiders. As a result the solutions above are not longer needed.