moveit/moveit #1928: Moving MoveIt Msgs into ros-planning/MoveIt

mlautman, 2020-03-04:
@henningkayser @rhaschke and @v4hn I am particularly interested to get your opinions here. I already know @davetcoleman's view on this :)

I am sure this has already been debated elsewhere but I wanted to (re)raise the issue because it seems clear to me that we should unify ros-planning/moveit and ros-planning/moveit_msgs into the same repo. Doing so will ensure that users always have the correct version of moveit_msgs for their checkout of MoveIt, a recurring issue (https://github.com/ros-planning/moveit/issues/1186, https://github.com/ros-planning/moveit/issues/1914) and make it easier for developers to make necessary changes to the messages without having to synchronize multiple PR's, another common pain point (eg: https://github.com/ros-planning/moveit_msgs/pull/65)

rhaschke, 2020-03-05:
I support this merge as it will facilitate synchronization as you said. The only downside I see is that it becomes harder to just checkout and build the messages.

gavanderhoorn, 2020-03-05:
I'll be the grumpy grand-dad who has to complain about everything, but when do we stop merging everything MoveIt related into `ros-planning/moveit`?

The Panda MoveIt config not being merged also causes headaches (see ROS Answers). FCL is also not aligning its development with MoveIt, with makes it really difficult to keep PRs in-sync with changes there. MTC?

v4hn, 2020-03-05:
My first reaction to this was:

*Never ever merge messages and code. It always makes people grumpy in integration projects.*

But thinking about MoveIt, we will of course keep the packages separate and there is always the possibility to use the released/binary version of `moveit_msgs` if you do not want to have the whole stack around.
The only case where it might be problematic would be unreleased changes to the messages that are required by external packages not using MoveIt.
Given that the package is called `**moveit**_msgs`, and not, e.g., `trajectory_msgs`, I agree this use-case is negligable and people who are affected clearly know how to handle it.

So while my gut feeling tells me to keep them separate, I explicitly do not vote either way.

felixvd, 2020-03-05:
Is there a way to maintain a separate moveit_msgs repository that blindly mirrors the state of the folder in the moveit repository?

tylerjw, 2020-03-11:
I don't understand the usecase where having them separate is better.  Is anyone really using moveit_msgs without moveit?

Having them separate makes it really annoying when you want to get a change merged from master in moveit to melodic-devel but you can't because that change depends on a change in moveit_msgs that is in master but not melodic-devel on that repo yet.

v4hn, 2020-03-11:
> I don't understand the usecase where having them separate is better.
> Is anyone really using moveit_msgs without moveit?

This is why I said this use-case (if it exists) is relevant for people with a lot of experience only.

> Having them separate makes it really annoying when you want to get a change merged from master in moveit
> to melodic-devel but you can't because that change depends on a change in moveit_msgs that is in master
> but not melodic-devel on that repo yet.

This makes it a two-step procedure right now. I would not call that "really annoying" yet.
Also I am not sure we still allow message changes in the melodic branch.
At least Dave(?) put this on the website:

> Melodic LTS: This LTS is stable and no API changes are accepted

v4hn, 2020-03-12:
>  I don't understand the usecase where having them separate is better.
> Is anyone really using moveit_msgs without moveit?

A very fresh example of a package that does this is https://github.com/tork-a/rqt_joint_trajectory_plot , since they added support to plot `DisplayRobotTrajectory` messages the other day.
Of course, the standard use-case would also have MoveIt running, but standalone trajectory analysis from rosbags is clearly feasible.

davetcoleman, 2020-06-02:
> I don't understand the usecase where having them separate is better. Is anyone really using moveit_msgs without moveit?

Anyone using MoveIt externally will just be using the debian release of moveit_msgs, not downloading it from source. For example, how often do you ``git clone`` [std_msgs](http://wiki.ros.org/std_msgs)? You don't.

> At least Dave(?) put this on the website

Yes, I thought we all agreed that was the meaning of a now 2 year old LTS distro. 

> This makes it a two-step procedure right now. I would not call that "really annoying" yet.

There are 10+ other repos in our MoveIt project, some that use moveit_msgs. So this does quickly get complex and annoying.

v4hn, 2020-06-02:
> There are 10+ other repos in our MoveIt project, some that use moveit_msgs. So this does quickly get complex and annoying.

I don't see how this is an argument as you would still depend on a pull-request for the main repo from these other projects...

Anyway, I thought the outcome of this discussion is that we could do it with almost no downside.