moveit/moveit #1931: Implementation of parameter "TranslationXY2D" IKFast

edetleon, 2020-03-05:
Hello,

I use ROS melodic with Ubuntu 18.04.
I had an issue with the IKFast (wich uses openrave 0.9.0) while generating the IK solver type TranslationXY2D for a robot arm that I created myself.*
The generation had some errors, but I could generate the package and compile my workspace anyway. When I opened it in Rviz (with the command roslaunch pkg_name demo.launch) and tried to move it, I couldn't move the robot and had the following error :

[ERROR] [1583159947.486174117]: IK for this IkParameterizationType not implemented yet.

This is exactly the same problem reported in the 2 links below for the IK solver type "Translation3D" and "TranslationYaxis4D", except that these posts concern older versions :

https://answers.ros.org/question/191884/ik-for-this-ikparameterizationtype-not-implemented-yet/
#548

I followed the developers advices and I added 2 lines of code in the source file "pkg_name_ikfast_moveit_plugin.cpp" generated by the solver. Now it works and I can move the arm with Rviz, using the interractive marker. This is the lines that I added after "case IKP_TranslationXY2D" (**line 592, you will fid the code below) :

trans[2] = 0;

  ComputeIk(trans, direction.data, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
  return solutions.GetNumSolutions();
It would be good to implement the TranslationXY2D type of IK solver, as well as the Translation2DOrientation1D type, because as far as I know, there is no existing IK solver for robots with low DOF.

If someone could read this, and take the time to do the implementation, thank you very much !

Best regards,

Edith

*To install and generate the IK Solver, I followed the tutorial linked below, and I used the following command line to generate the solver :
rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype TranslationXY2D $MYROBOT_NAME.urdf <planning_group_name> <base_link> <eef_link>

http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html

**Here is the code concerned :

`solutions.Clear();

double trans[3];
trans[0] = pose_frame.p[0];
trans[1] = pose_frame.p[1];
trans[2] = pose_frame.p[2];

KDL::Rotation mult;
KDL::Vector direction;

switch (GetIkType())
{
case IKP_Transform6D:
case IKP_Translation3D:

  mult = pose_frame.M;

  double vals[9];
  vals[0] = mult(0, 0);
  vals[1] = mult(0, 1);
  vals[2] = mult(0, 2);
  vals[3] = mult(1, 0);
  vals[4] = mult(1, 1);
  vals[5] = mult(1, 2);
  vals[6] = mult(2, 0);
  vals[7] = mult(2, 1);
  vals[8] = mult(2, 2);

  // IKFast56/61
  ComputeIk(trans, vals, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
  return solutions.GetNumSolutions();

case IKP_Direction3D:
case IKP_Ray4D:
case IKP_TranslationDirection5D:
  // For **Direction3D**, **Ray4D**, and **TranslationDirection5D**, the first 3 values represent the target
  // direction.

  direction = pose_frame.M * KDL::Vector(0, 0, 1);
  ComputeIk(trans, direction.data, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
  return solutions.GetNumSolutions();

case IKP_TranslationXAxisAngle4D:
case IKP_TranslationYAxisAngle4D:
case IKP_TranslationZAxisAngle4D:
  // For *TranslationXAxisAngle4D*, *TranslationYAxisAngle4D*, *TranslationZAxisAngle4D* - end effector origin
  // reaches desired 3D translation, manipulator direction makes a specific angle with x/y/z-axis (defined in the
  // manipulator base link’s coordinate system)
  ROS_ERROR_NAMED(name_, "IK for this IkParameterizationType not implemented yet.");
  return 0;

case IKP_TranslationLocalGlobal6D:
  // For **TranslationLocalGlobal6D**, the diagonal elements ([0],[4],[8]) are the local translation inside the end
  // effector coordinate system.
  ROS_ERROR_NAMED(name_, "IK for this IkParameterizationType not implemented yet.");
  return 0;

case IKP_Rotation3D:
case IKP_Lookat3D:
case IKP_TranslationXY2D:

  trans[2] = 0;

  ComputeIk(trans, direction.data, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
  return solutions.GetNumSolutions();

case IKP_TranslationXYOrientation3D:
  ROS_ERROR_NAMED(name_, "IK for this IkParameterizationType not implemented yet.");
  return 0;

case IKP_TranslationXAxisAngleZNorm4D:
  double roll, pitch, yaw;
  // For **TranslationXAxisAngleZNorm4D** - end effector origin reaches desired 3D translation, manipulator
  // direction needs to be orthogonal to z axis and be rotated at a certain angle starting from the x axis (defined
  // in the manipulator base link’s coordinate system)
  pose_frame.M.GetRPY(roll, pitch, yaw);
  ComputeIk(trans, &yaw, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
  return solutions.GetNumSolutions();

case IKP_TranslationYAxisAngleXNorm4D:
  // For **TranslationYAxisAngleXNorm4D** - end effector origin reaches desired 3D translation, manipulator
  // direction needs to be orthogonal to x axis and be rotated at a certain angle starting from the y axis (defined
  // in the manipulator base link’s coordinate system)
  pose_frame.M.GetRPY(roll, pitch, yaw);
  ComputeIk(trans, &roll, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
  return solutions.GetNumSolutions();

case IKP_TranslationZAxisAngleYNorm4D:
  // For **TranslationZAxisAngleYNorm4D** - end effector origin reaches desired 3D translation, manipulator
  // direction needs to be orthogonal to y axis and be rotated at a certain angle starting from the z axis (defined
  // in the manipulator base link’s coordinate system)
  pose_frame.M.GetRPY(roll, pitch, yaw);
  ComputeIk(trans, &pitch, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
  return solutions.GetNumSolutions();

default:
  ROS_ERROR_NAMED(name_, "Unknown IkParameterizationType! "
                         "Was the solver generated with an incompatible version of Openrave?");
  return 0;
}
}

void IKFastKinematicsPlugin::getSolution(const IkSolutionList& solutions, int i,
std::vector& solution) const
{
solution.clear();
solution.resize(num_joints_);

// IKFast56/61
const IkSolutionBase& sol = solutions.GetSolution(i);
std::vector vsolfree(sol.GetFree().size());
sol.GetSolution(&solution[0], vsolfree.size() > 0 ? &vsolfree[0] : nullptr);

for (std::size_t joint_id = 0; joint_id < num_joints_; ++joint_id)
{
if (joint_has_limits_vector_[joint_id])
{
solution[joint_id] = enforceLimits(solution[joint_id], joint_min_vector_[joint_id], joint_max_vector_[joint_id]);
}
}
}`

rhaschke, 2020-03-06:
If there are just a few code lines to add, please do so in this file:
https://github.com/ros-planning/moveit/blob/fd213f43f0451ee54e128fb421f53dbef480fc75/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L607
and file a PR. Thanks.

edetleon, 2020-03-06:
> If there are just a few code lines to add, please do so in this file:
> https://github.com/ros-planning/moveit/blob/fd213f43f0451ee54e128fb421f53dbef480fc75/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L607
> 
> 
> and file a PR. Thanks.
> Note that these changes should be made to the `master` branch.

Thank you for your answer. I don't really know what I should add, as I am not so familiar with coding..

rhaschke, 2020-03-06:
I'm confused, because you originally wrote:

> I followed the developers advices and I added 2 lines of code in the source file "pkg_name_ikfast_moveit_plugin.cpp" generated by the solver. Now it works and I can move the arm with Rviz, using the interractive marker. This is the lines that I added after "case IKP_TranslationXY2D" (**line 592, you will find the code below) :

```c++
trans[2] = 0;
ComputeIk(trans, direction.data, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
return solutions.GetNumSolutions();
```
Hence, you exactly know what to do. I'm just asking to add these lines to the above mentioned file and report on the results.

v4hn, 2020-03-20:
Resolved by #1949. Thanks again for the patch.

closed 2020-03-20 by v4hn