moveit/moveit #1933: To support collision_detection_bullet in kinetic branch

Hubert51, 2020-03-06:
I want to use collision_detection_bullet functionality in the kinetic branch. Currently, it is only supported in the master branch. I am trying to merge them. I commented out several lines to make it compilable. These codes are in the `moveit_core/collision_detection/src/world.cpp` and `moveit_core/robot_state/src/conversion.cpp`. Both of the codes are related to transformation. May I ask do you have plan to make collision_detection_bullet workable in the kinetic branch?  

Thank you!

rhaschke, 2020-03-06:
If you want to use the master-branch features in Kinetic, we suggest to compile the master branch from source. This also works on Kinetic. We will not backport those changes to Kinetic.

closed 2020-03-06 by rhaschke

Hubert51, 2020-03-06:
I added the master branch moveit package in my src folder.  
When I compile the master branch and try to run the `test_bullet_continuous_collision_collision_checking.cpp`.   
The compiler gave me this error:  `/home/ruijie/kinova_ws/devel/lib/libmoveit_collision_detection_bullet.so.1.0.1: undefined reference to "btPolyhedralConvexShape::setPolyhedralFeatures(btConvexPolyhedron&)"`