moveit/moveit #1938: To use collision_detection_bullet in kinetic branch

Hubert51, 2020-03-06:
I have an issue in https://github.com/ros-planning/moveit/issues/1933#issue-576739605
I added the master branch moveit package in my src folder.  
When I compile the master branch and try to run the `test_bullet_continuous_collision_collision_checking.cpp`.   
The compiler gave me this error:  `/home/ruijie/kinova_ws/devel/lib/libmoveit_collision_detection_bullet.so.1.0.1: undefined reference to "btPolyhedralConvexShape::setPolyhedralFeatures(btConvexPolyhedron&)"`

rhaschke, 2020-03-06:
@j-petit can you comment on this? Why are there symbols missing when linking the bullet collision detector on Kinetic?

Hubert51, 2020-03-06:
I used catkin_make to build source files. And it gave me an error that I need to use catkin_make_isolated to build them. So I deleted several folders such as `moveit/moveit_ros/moveit_ros/` and `moveit/moveit_plannar/moveit_plannar` and then I can use catkin_make to compile the project.  

Today, I tried to use catkin_make_isolated to build moveit master branch and got the following error:
```
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_vector_container.h:40,
                 from /opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_containers.h:40,
                 from /opt/ros/kinetic/include/geometric_shapes/bodies.h:45,
                 from /home/ruijie/kinova_ws/src/moveit/moveit_core/collision_distance_field/include/moveit/collision_distance_field/collision_distance_field_types.h:47,
                 from /home/ruijie/kinova_ws/src/moveit/moveit_core/collision_distance_field/src/collision_distance_field_types.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/ruijie/kinova_ws/src/moveit/moveit_core/collision_distance_field/src/collision_distance_field_types.cpp:55:22:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/build.make:62: recipe for target 'collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/src/collision_distance_field_types.cpp.o' failed
```
Would you to have a try and tell me how to compile moveit-master in kinetic and can successfully run `test_bullet_continuous_collision_collision_checking.cpp` file. 

Thank you very much!

Hubert51, 2020-03-06:
I tried to reinstall the bullet-2.87 and compile the project again. Now It is compiled!  

But when I was running the `test_bullet_continuous_collision_collision_checking.cpp`, it gave me the error looks like a memory bug:
```
ruijie@ubuntu:~$ rosrun moveit_core test_bullet_continuous_collision_checking
[==========] Running 3 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 1 test from BulletCollisionDetectionTester
[ RUN      ] BulletCollisionDetectionTester.ContinuousCollisionWorld
[ INFO] [1583538609.420898321]: Loading robot model 'panda'...
*** Error in `/home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking': malloc(): memory corruption: 0x0000000001ce46e0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f337cc327e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8213e)[0x7f337cc3d13e]
/lib/x86_64-linux-gnu/libc.so.6(__libc_malloc+0x54)[0x7f337cc3f184]
/usr/lib/x86_64-linux-gnu/libLinearMath.so.2.83(+0x36d3)[0x7f337d5206d3]
/usr/lib/x86_64-linux-gnu/libBulletCollision.so.2.83(_ZN16btDbvtBroadphaseC2EP22btOverlappingPairCache+0x10f)[0x7f337bb519ef]
/home/ruijie/kinova_ws/devel/lib/libmoveit_collision_detection_bullet.so.1.0.1(_ZN26collision_detection_bullet16BulletBVHManagerC1Ev+0x184)[0x7f337f0149f4]
/home/ruijie/kinova_ws/devel/lib/libmoveit_collision_detection_bullet.so.1.0.1(_ZN26collision_detection_bullet24BulletDiscreteBVHManagerC1Ev+0x18)[0x7f337effecf4]
/home/ruijie/kinova_ws/devel/lib/libmoveit_collision_detection_bullet.so.1.0.1(_ZN19collision_detection18CollisionEnvBulletC1ERKSt10shared_ptrIKN6moveit4core10RobotModelEEdd+0xbc)[0x7f337f008d9a]
/home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking(_ZN30BulletCollisionDetectionTester5SetUpEv+0x227)[0x43743d]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8internal38HandleSehExceptionsInMethodIfSupportedINS_4TestEvEET0_PT_MS4_FS3_vEPKc+0x65)[0x7f337f568046]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8internal35HandleExceptionsInMethodIfSupportedINS_4TestEvEET0_PT_MS4_FS3_vEPKc+0x5a)[0x7f337f562adf]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing4Test3RunEv+0x7f)[0x7f337f546e0f]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8TestInfo3RunEv+0x104)[0x7f337f547728]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8TestCase3RunEv+0x107)[0x7f337f547e49]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8internal12UnitTestImpl11RunAllTestsEv+0x2d2)[0x7f337f54f1a0]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8internal38HandleSehExceptionsInMethodIfSupportedINS0_12UnitTestImplEbEET0_PT_MS4_FS3_vEPKc+0x65)[0x7f337f569625]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8internal35HandleExceptionsInMethodIfSupportedINS0_12UnitTestImplEbEET0_PT_MS4_FS3_vEPKc+0x5a)[0x7f337f56398d]
/home/ruijie/kinova_ws/build/gtest/gtest/libgtest.so(_ZN7testing8UnitTest3RunEv+0xc6)[0x7f337f54dbe4]
/home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking(_Z13RUN_ALL_TESTSv+0x11)[0x435c6e]
/home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking(main+0x27)[0x435866]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f337cbdb830]
/home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking(_start+0x29)[0x431f69]
======= Memory map: ========
00400000-00451000 r-xp 00000000 07:00 6112935                            /home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking
00650000-00651000 r--p 00050000 07:00 6112935                            /home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking
00651000-00652000 rw-p 00051000 07:00 6112935                            /home/ruijie/kinova_ws/devel/lib/moveit_core/test_bullet_continuous_collision_checking
01954000-01d32000 rw-p 00000000 00:00 0                                  [heap]
7f3368000000-7f3368021000 rw-p 00000000 00:00 0 
7f3368021000-7f336c000000 ---p 00000000 00:00 0 
7f336fb46000-7f336fe77000 rw-p 00000000 00:00 0 
7f336fe77000-7f336fe82000 r-xp 00000000 07:00 4723757                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f336fe82000-7f3370081000 ---p 0000b000 07:00 4723757                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f3370081000-7f3370082000 r--p 0000a000 07:00 4723757                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f3370082000-7f3370083000 rw-p 0000b000 07:00 4723757                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f3370083000-7f3370089000 rw-p 00000000 00:00 0 
7f3370089000-7f3370094000 r-xp 00000000 07:00 4723765                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
7f3370094000-7f3370293000 ---p 0000b000 07:00 4723765                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
7f3370293000-7f3370294000 r--p 0000a000 07:00 4723765                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
7f3370294000-7f3370295000 rw-p 0000b000 07:00 4723765                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
7f3370295000-7f33702ab000 r-xp 00000000 07:00 4723754                    /lib/x86_64-linux-gnu/libnsl-2.23.so
7f33702ab000-7f33704aa000 ---p 00016000 07:00 4723754                    /lib/x86_64-linux-gnu/libnsl-2.23.so
7f33704aa000-7f33704ab000 r--p 00015000 07:00 4723754                    /lib/x86_64-linux-gnu/libnsl-2.23.so
[... cut ...]
7f337a420000-7f337a421000 rw-p 00059000 07:00 8141851                    /usr/lib/x86_64-linux-gnu/libqhull.so.7.2.0
7f337a421000-7f337a9e3000 r-xp 00000000 07:00 8140950                    /usr/lib/x86_64-linux-gnu/libassimp.so.3.2.0
7f337a9e3000-7f337abe3000 ---p 005c2000 07:00 8140950                    /usr/lib/x86_64-linux-gnu/libassimp.so.3.2.0
7f337abe3000-7f337aca9000 r--p 005c2000 07:00 8140950                    /usr/lib/x86_64-linux-gnu/libassimp.so.3.2.0
7f337aca9000-7f337acab000 rw-p 00688000 07:00 8140950                    /usr/lib/x86_64-linux-gnu/libassimp.so.3.2.0Aborted (core dumped)

```

j-petit, 2020-03-07:
Which version of Ubuntu are you using?

rhaschke, 2020-03-07:
You should not have deleted selected MoveIt folders. This essentially mixes up some new files from master branch with some old files from binary packages.
Please just follow the instructions from the [tutorials](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html#create-a-catkin-workspace-and-download-moveit-source). This states to fetch several source packages via `wstool`.

Hubert51, 2020-03-09:
Thanks! I will try again from the starting point. Currently I am using ubuntu 16.04 and master branch of the moveit