moveit/moveit #1939: Goal state in Rviz should not be set to "current" if execution fails

felixvd, 2020-03-07:
I was playing around in the tutorial and got into a situation where the planning Start State was not near the "real" robot state. Executing the path failed, but the Goal State was still reset to the current robot state. I think this is unintuitive and unhelpful - I am not sure why the Goal State needs to be reset at all, but at least it shouldn't happen when the execution fails, I believe. Usually you would fix the issue and then attempt to plan and execute again.

rhaschke, 2020-03-08:
I cannot reproduce the described behavior. Using the default settings, I can only confirm that the Start State is reset to current (as indicated by the corresponding combobox).

martiniil, 2020-07-30:
Hi.

> got into a situation where the planning Start State was not near the "real" robot state.

This happened to me, too. How is this even possible?

Besides, I cannot reproduce the described behavior about the goal state in this situation. After the path failed I can execute it successfully by clicking plan and execute again.