moveit/moveit #1940: Rviz crashes frequently if Show Trail and Loop Animation are used together

felixvd, 2020-03-07:
### Description

When displaying a plan with the MotionPlanning plugin, Rviz tends to crash if both `Loop Animation` and `Show Trail` (in the `Planned Path` category) are turned on while the animation is playing.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build

### Steps to reproduce
Plan a path (optionally: execute it), turn on "Loop Animation" and then turn on "Show Trail". If that doesn't work, turn them both on/off a few times. I could reproduce it fairly reliably this way.

### Expected behaviour
Rviz shouldn't crash.

### Actual behaviour
Rviz crashes

### Backtrace or Console output

```
[rviz_acd9b44678e6_219_7606801427590083783-7] process has died [pid 289, exit code -11, cmd /opt/ros/melodic/lib/rviz/rviz -d /opt/ros/melodic/share/panda_moveit_config/launch/moveit_empty.rviz __name:=rviz_acd9b44678e6_219_7606801427590083783 __log:=/root/.ros/log/2c6030e6-606f-11ea-b8eb-0242ac110002/rviz_acd9b44678e6_219_7606801427590083783-7.log].
log file: /root/.ros/log/2c6030e6-606f-11ea-b8eb-0242ac110002/rviz_acd9b44678e6_219_7606801427590083783-7*.log

```

Dale-Koenig, 2020-03-07:
Played around this for an hour or so.  It seems this (and some related things) are solved by removing the call to `QApplication::processEvents();` from `RobotStateVisualization::load`

As it is, there can be bad unintented recursion where `TrajectoryVisualization::update` calls `TrajectoryVisualization::changedShowTrail` calls `RobotStateVisualization::load` calls `QApplication::processEvents` which again leads to a call to `TrajectoryVisualization::update` at which point very bad things happen.

rhaschke, 2020-03-08:
Fixed via #1941.

closed 2020-03-08 by rhaschke

felixvd, 2020-03-09:
Thanks, this went at record speed :)